bambulab/BambuStudio #6649: H2D loses nozzle pressure

Farleken, 2025-04-22:
### Bambu Studio Version

v02.00.02.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

H2D

### How to reproduce

Ams 2 pro Autorefill function

2 identical filaments for auto-refill
1. Spool empty
2. Prints until the valid stop
3. Moves to the cleaning station
4. Rinses out old filament
5. Requests a new spool
6. Cleans old filament -> new spool
7. Cleans nozzle
8. Moves to the object and continues printing
9. Starts at the same position, but loses nozzle pressure

### Actual results

When two identical filaments are running in automatic refill mode,
one empty spool is replaced by a new full spool (A1->A2)
and moves to the designated print position.
The filament flow is incorrect.

For example, if the filament touches the outer wall, the filament will not feed properly.

[image]

### Expected results

The nozzle should move to a position where it is not obvious that a filament refill has taken place.

E.g., filament flushed, old -> new -> to object -> infill -> actual finished printing position.

To fill the nozzle pressure.

### Project file & Debug log uploads

[Noozle_Brush3mf_.zip](https://github.com/user-attachments/files/19856170/Noozle_Brush3mf_.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gaolegao-lx, 2025-04-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Farleken, 2025-04-24:
I can't upload a zip file anymore. I keep getting errors.

The ticket has been sent. Ticket No.: US250424556004

BoBoYu-BamBu, 2025-04-25:
@Farleken Thank you. This is a problem on the firmware side. Please follow the instructions provided above to create a post-sale order on Handy, which will help solve the problem more quickly.

realzorro, 2025-04-27:
We attach great importance to your feedback.
This issue requires uploading printer logs and 3MF files. You can create a technical ticket on the official website to have our technical experts assist you.
Link to create technical work ticket: https://bambulab.com/en/my/support/tickets/create
Looking forward to receiving your ticket as soon as possible.

Farleken, 2025-04-27:
The ticket has already been submitted and is being processed.

Thank you

closed 2025-04-28 by BoBoYu-BamBu