bambulab/BambuStudio #6650: When I updated bambu studio it didn't work again

abdullah7447, 2025-04-22:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

W:10

### Additional system information

G

### Printer

P1S

### How to reproduce

[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-23:
Sorry, could we try another network instead? If it still doesn't work, please let us know.

BoBoYu-BamBu, 2025-04-23:
@abdullah7447 Could you provide us with the log files for our analysis?

BoBoYu-BamBu, 2025-04-23:
@abdullah7447  Please provide the following information: Could you please provide the log files for us to analyze? Was the previous version working properly? What is the language of Studio?

abdullah7447, 2025-04-23:
@BoBoYu-BamBu The previous version worked fine and the Bamboo Studio language was English, How can I give you the log files?

BoBoYu-BamBu, 2025-04-24:
@abdullah7447 
As shown in the figure below, click on the drop-down box of the file, select "Help", choose "Show Configuration Folder" to open it, find the log file, and then send it to me for all the contents to be packed.
[image]
[image]

abdullah7447, 2025-04-24:
[log_iotc.txt](https://github.com/user-attachments/files/19880987/log_iotc.txt)
@BoBoYu-BamBu

BoBoYu-BamBu, 2025-04-24:
@abdullah7447 Thanks，Isn't there anything more complete? There is no useful information at all.

abdullah7447, 2025-04-24:
@BoBoYu-BamBu  What am I supposed to do?

BoBoYu-BamBu, 2025-04-24:
@abdullah7447  According to the operation shown in the attached figure, please pack all the files in the log directory for me. Thank you.
[image]
[image]

abdullah7447, 2025-04-24:
[log_iotc.txt](https://github.com/user-attachments/files/19881133/log_iotc.txt)

@BoBoYu-BamBu

abdullah7447, 2025-04-24:
[log_iotc.txt](https://github.com/user-attachments/files/19881160/log_iotc.txt)

@BoBoYu-BamBu

BoBoYu-BamBu, 2025-04-24:
@abdullah7447 Thank you. Is that all?

abdullah7447, 2025-04-24:
yes @BoBoYu-BamBu

abdullah7447, 2025-04-25:
@BoBoYu-BamBu Is there any update?

abdullah7447, 2025-04-27:
@BoBoYu-BamBu  ؟

realzorro, 2025-04-28:
You can try the following method: change the language to another language in the configuration interface, and then switch back to English to see if it can solve your problem

abdullah7447, 2025-04-28:
@realzorro How can I do that?

realzorro, 2025-04-28:
> [@realzorro](https://github.com/realzorro) How can I do that?

Changed the language in Bambu Studio. Preference > Language. 
Change the language to another language, and then change back to English to see if it can solve your problem

abdullah7447, 2025-04-28:
Thank you so much it worked @realzorro