bambulab/BambuStudio #6653: Gryoid Infill Prints Runs Slower With High Flow Nozzles than Standard Flow Nozzles With PETG HF

hifihedgehog, 2025-04-23:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

CPU: AMD Ryzen 9 7950X
Memory: 32GB
Display Card: NVIDIA GeForce RTX 3080

### Printer

Bambu Lab H2D

### How to reproduce

1. Choose any model (of those I have tested so far, all appear to exhibit this behavior). A primitive object like a cube, scaled up to 200mm in x, y, z, will do.

2. Change sparse infill pattern to gyroid.

3. Ensure you have Bambu Lab PETG HF selected. 

4. Slice with first both nozzles set to standard and then high flow for flow. Compare print times.

### Actual results

[image]

[image]

### Expected results

Print times should be faster.

### Project file & Debug log uploads

[200mmx200mmx200mm Cube, PETG HF, Gryoid Infill - High Flow Nozzles.zip](https://github.com/user-attachments/files/19860163/200mmx200mmx200mm.Cube.PETG.HF.Gryoid.Infill.-.High.Flow.Nozzles.zip)
[200mmx200mmx200mm Cube, PETG HF, Gryoid Infill - Standard Flow Nozzles.zip](https://github.com/user-attachments/files/19860164/200mmx200mmx200mm.Cube.PETG.HF.Gryoid.Infill.-.Standard.Flow.Nozzles.zip)
[debug_Tue_Apr_22_22_29_44_29972.log.0.zip](https://github.com/user-attachments/files/19860186/debug_Tue_Apr_22_22_29_44_29972.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

RickReeser, 2025-05-21:
Did a bit of testing. Lowering the max volumetric speed causes the time estimate to go down in some cases.
Here's some results from my tests. The only difference between each slice was the max volumetric speed in the filament profile. All of them were done on left HF nozzle on H2D.

| Max flow | Sparse infill print time |
|----------|--------------------------|
| 6        | 4h 55m                   |
| 8        | 4h 20m                   |
| 10       | 4h 16m                   |
| 12       | 4h 30m                   |
| 14       | 4h 38m                   |
| 20       | 4h 47m                   |
| 30       | 4h 53m                   |

[gyroid slice test.zip](https://github.com/user-attachments/files/20359775/gyroid.slice.test.zip)

isaaccitrom-oss, 2026-09-01:
Interesting, I am currently running a large print with gyroid sparse infill. My X2D is printing the gyroid at about 50 mm/s to 80 mm/s, I estimate. I can't figure it out. By the way, I am not printing PETG HF. I am printing PLA in this instance. Very odd.  20 mm cubed/sec maximum volumetric speed.