bambulab/BambuStudio #6654: The Multi-device tab no longer shows the print progress bar

luder888, 2025-04-23:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

In Bambu Studio 1.X version, while printing, the "Multi-device" tab used to show a green progress bar showing the remaining print time, similar to how the "Device" tab's progress bar.

[image]

### Actual results

The "Multi-device" tab no longer shows a print progress bar. Instead, it simply displays a text showing the current status such as Printing or Idle.

### Expected results

 The "Multi-device" tab should show a green progress bar showing the remaining print time, similar to how the "Device" tab's progress bar. This feature existed in the 1.X version. This is extremely useful if you have multiple printers and be able to see the progress for all printers in one screen.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

frantiseklorenc, 2025-05-20:
Hello, this bug is very annoying when you have multiple devices. It would be great to have fix for this bug in next release, it's been a month now. Thanks

frantiseklorenc, 2025-07-28:
Yes, it works. Thanks.