bambulab/BambuStudio #6656: Ok, what am I doing wrong.  Change of nozzle causes single layer bottom layer to not slice correctly

rjnerd, 2025-04-23:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel laptop, about 6 years old, has a 1.8 ghz I7 laptop processor, 16gb ram (LG brand)

### Printer

Bambu h2d

### How to reproduce

Use the makerword niswk FPV-Wing-Racer-Rev2
move everything onto one sheet., (lateral moves only for the wing assemblies)
Slice for petg-hf 0.4mm nozzle..  Notice that the wings have a big area of single layer skin, as do the spare ailerons, and wingtips.

change to 0.6 nozzle. leave the rest alone.

Slice model.  Notice, while there is a large area of wing skin in the color of your choice, post slicing, there is no actual filament to be deposited in the wing skin, just the rib skeleton.   However, the wing tips and the spare ailerons (which have edges on all sides, unlike the wing skin that has the single layer just ending, do get a deposit of filament.)

I did try a few things to better force the issue, including requesting ironing, setting a minimum thickness for the bottom shell layer. (both changing the measured thickness to other than 0), and playing around with thinner layers via picking a different @BBL process.

### Actual results

the printer doesn't deposit any filament to the wing skin when using .6 nozzle. 

### Expected results

Filament applied to wing skin, same as the .4 mm nozzle version.

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-23:
Thank you for your feedback. Could you please provide us with screenshots and log files?

rjnerd, 2025-04-25:
OK, here are screenshots.  Didn't send it to the printer, so no log.
(unless there is a slicer log, if so, tell me how to find it)
-dp-
Celebrate Susan's life 1956-2020
<https://www.lastingmemories.com/memorial/susan-fera-assmann>
Founder: The New England Rubbish Deconstruction Society
<http://the-nerds.org>

This planet needs a lot more kids who think taking a lawnmower apart is
more fun than playing a videogame