bambulab/BambuStudio #6658: Unreasonable support surface angles

AnalogKnight, 2025-04-23:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

H2D

### How to reproduce

1. Create supports, making sure the number of support layers is singular.
2. Set the number of layers of support contact surface to 2(or any even number).
3. Start slicing

### Actual results

[image]

[image]
The bridge direction doesn't make sense, it should be towards the vertical rather than the horizontal so that less overhang can go through. This causes the bridge to collapse easily

### Expected results

The direction of the bridge should cross the body of the support and not be the same direction.
If the number of support surface layers is changed to 3, the direction will be reasonable.

[image]
There should be a parameter for modifying the orientation of the support bridge, or making the support generation smarter.

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-23:
Thank you for your feedback. Could you please provide the model file?

AnalogKnight, 2025-04-23:
> Thank you for your feedback. Could you please provide the model file?

Thanks for the response.

[test.zip](https://github.com/user-attachments/files/19873053/test.zip)
I found that it didn't show up at all times, but I tried to reproduce it.

NothFie, 2025-04-24:
@AnalogKnight Thanks for your feedback. The angle is indeed a bug and we are fixing it to make sure the lowest layer interlace with support base. For now you could set the INTERFACE PATTERN to RECTILINEAR to avoid this bug.
In your situation, we do not recommend to use support materials for support base (in future versions there will be tips to explain that). And now according to the program settings, when the interface material differs from the base material, we have added transition layers. This enhancement improves the bridging quality at the interface and ensures proper angular intersections. However, if you set the base material to also be the support material, this functionality will be bypassed.

AnalogKnight, 2025-04-24:
Thank you for your advice.
I don't usually use the support material for base, but sometimes I have to. For example when there is a very narrow and deep depression on the model, if I use the support material only for the interface, the interface will always stick to the model when I remove the support. If the support material is also used for the base, I can remove them intact.

NothFie, 2025-05-12:
The bug is fixed and will be released in the next vesion.