bambulab/BambuStudio #6659: Bambulabs Studion Always on top

Ihasabucket8950, 2025-04-23:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 24H2

### Additional system information

Computer: Acer Nitro ANV16-41
CPU: AMD Ryzen 7 8845HS
Memory: 32 GB DDR5 SODIMM
GPU: NIVIDIA GeForce RTX 4060 Laptop GPU
GPU: AMD Radeon 780M Graphics

### Printer

Bambu Lab A1

### How to reproduce

Open Bambu Studio...
Click Maximize...
Try opening the taskbar by moving mouse to the taskbar (Hide Taskbar Must Be On)...

### Actual results

https://github.com/user-attachments/assets/694f53f1-a55e-4d53-826a-9dccc0c21e8a
Does not Open Taskbar without pressing restore or the windows key.

### Expected results

Should be able to open the Taskbar without pressing restore or the windows key all while maximized.

### Project file & Debug log uploads

No need for project file this has done this for multiple updates.

[debug_Tue_Apr_22_21_51_58_14248.log](https://github.com/user-attachments/files/19862038/debug_Tue_Apr_22_21_51_58_14248.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-04-23:
Okay, we have confirmed that the issue is that the taskbar on Windows cannot reappear after being automatically hidden.
Currently, you can press the key 'win' between Ctrl and Alt to trigger the display of the taskbar

Haidiye00, 2025-04-23:
Okay, the problem has been identified and will be resolved

JonLaliberte, 2025-05-19:
There is a similar problem on macOS. The app doesn't obey properly when told to hide.

For example, if I go to the Finder and go to the Finder menu and go to Hide Others, some instances of Bambu Studio just refuse to hide. To get it to work I have to click on the BS window, then I can hide it.

Haidiye00, 2025-06-03:
Hello, please use the latest version of the software. Thank you~
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59

closed 2025-06-03 by Haidiye00