bambulab/BambuStudio #6660: Login bug.

marc509d, 2025-04-23:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

i can't login, i click the button and then a white box appears and it doesn't take me to the login page.

### Printer

Bambu lab A1

### How to reproduce

go to the main page.
click Login/Register.
and that's it...

### Actual results

[image]

[image]

### Expected results

Normally the login/register page would appear so i can login.

### Project file & Debug log uploads

it aint the projects messing with me.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-23:
@marc509d could we try another network instead? Please provide the following information: Could you please provide the log files for us to analyze? Was the previous version working properly? What is the language of Studio?

marc509d, 2025-04-23:
how do i provide the log files? and one previous version was working but not any of the recent ones, and the studio launguage is danish

marc509d, 2025-04-23:
@BoBoYu-BamBu it fixed itself somehow. idk.

BoBoYu-BamBu, 2025-04-23:
@marc509d So, is the situation okay now?

marc509d, 2025-04-23:
@BoBoYu-BamBu yeah, ig.

closed 2025-04-23 by BoBoYu-BamBu