bambulab/BambuStudio #6661: Double modle on build plate

ashelysium12, 2025-04-23:
### Bambu Studio Version

Latest one 

### Where is the application from?

Bambu Lab Official website

### OS version

Latest One

### Printer

Bambu Lab a1 and a1 mini

### How to reproduce

If i click on the recent opened models it brings it a new project with the original model still there

### Actual results

2 models 1 on top of the other one 

### Expected results

only have 1 model there

### Project file & Debug log uploads

Dont have one 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-24:
Hello, we have fixed the double-click issue and it will be included in the next version. Currently, you can click, do not double-click

ashelysium12, 2025-04-24:
Thank you Sent from my iPhone

Haidiye00, 2025-04-27:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

closed 2025-04-27 by Haidiye00