bambulab/BambuStudio #6674: Ams lite is not recognized

Corwold, 2025-04-24:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

A1 Mini

### How to reproduce

Set Multi Color for the Model
Then Slice it
Then Print and set A1 and A2 (but in Print Menu the Button "Use AMS" is missing)
Print the Model 

### Actual results

In Field Printer where you can look for Process and AMS Lite and External Spool the Print Jumps to External Spool when Filament is Loaded in AMS
If no Filament is Loaded the Printer Says no Filament in external Spool. 
In Version1.9.7.50 all is working fine. Here is also the Button "Use AMS" in Print Menu. If no Filament is Loaded the Print use the AMS and Load A1 how its Set and makes the Colorchange at the Point where he Should

[image] This is Prin Menu in 2.0.2.57

[image] an This is Print Menu in 1.9.7.50

### Expected results

Color change and use of AMS Lite

### Project file & Debug log uploads

[JESUS KEYCHAIN.gcode.zip](https://github.com/user-attachments/files/19890407/JESUS.KEYCHAIN.gcode.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Corwold, 2025-04-24:
Hi thanks for Info. Now it can't load the Profile

[image]

Corwold, 2025-04-24:
Profile Loading has worked now. I test now the printing and let you know

Corwold, 2025-04-24:
First Print is done. Looks good. He makes the FilamentChange normlay. Thanks a lot to the Team for help and the Fix in the Software

BoBoYu-BamBu, 2025-04-25:
@Corwold Thank, Can this issue be closed now?

Corwold, 2025-04-25:
One last question. is it wanted that there is no longer a button for "Use AMS" ? Because the Checkbox for this is still not there. Is it now Automaticly used?

JohnSmithCYM, 2025-04-25:
Yes, the "Use AMS" have been removed from Studio V2.0.2 and the ext spool mapping is just 'like' what AMS do, it means you should make the filament type in slice file correspond to filament type in device ext setting.

closed 2025-04-25 by Corwold