bambulab/BambuStudio #6675: Bambu studio crashing during or just after Gcode export

Jameshigson-PD, 2025-04-24:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	Intel(R) Xeon(R) E-2276M  CPU @ 2.80GHz   2.81 GHz
Installed RAM	48.0 GB (47.7 GB usable)
System type	64-bit operating system, x64-based processor
Nvidia Quadro T2000

### Printer

H2D

### How to reproduce

Failure is happening in multiple scenarios, different step files, with and without other applications running. Immediately after restarting the laptop. Failures have become increasingly frequent since H2D software update. I didn't have a problem when slicing and printing on my X1C prior to the launch of H2D. 
1. launch BBS
2. sync printer
3. auto orientate 
4. slice

### Actual results

Software crash. Window closes and program stops running. Either while generating the G-code or just after. Crash has also happened just after slice completed. 

### Expected results

Slice complete > send to print
*note: I have no issues on the Mac version of the software. 

### Project file & Debug log uploads

[James Higson Crash File.zip](https://github.com/user-attachments/files/19891057/James.Higson.Crash.File.zip)

[bambu-studio.exe.9168.zip](https://github.com/user-attachments/files/19891075/bambu-studio.exe.9168.zip)

[log.zip](https://github.com/user-attachments/files/19891096/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-25:
Hello, may I ask if slicing directly without automatic orientation will still crash？
please try https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54
and provide new log and dmp

Jameshigson-PD, 2025-04-25:
Updated and still crashing. Do you want a crash log?

Haidiye00, 2025-04-27:
[image]
Hello, please provide us with the latest log folder and package it for us. thanks

Jameshigson-PD, 2025-04-28:
Please can you fix this bug? It's making the software completely useless because I can't send anything to the printers. 

[log (28.04.05).zip](https://github.com/user-attachments/files/19936709/log.28.04.05.zip)

Jameshigson-PD, 2025-04-28:
I'm using the latest version: V2.0.3.54

Haidiye00, 2025-04-28:
Okay, thank you for your support and feedback. We will analyze it as soon as possible~

Jameshigson-PD, 2025-04-29:
I'm attampting to print a different file now. It appears to slice ok and then when I click 'send to print' it crashes. 

[log (29.04.25).zip](https://github.com/user-attachments/files/19954962/log.29.04.25.zip)

Jameshigson-PD, 2025-04-30:
Another crash again this morning, the software is completely unusable. Please can you fix this urgently. 

[log (30.04.25).zip](https://github.com/user-attachments/files/19972933/log.30.04.25.zip)

guanyun-gudujian, 2025-04-30:
> Another crash again this morning, the software is completely unusable. Please can you fix this urgently, I use my printers for work.
> 
> [log (30.04.25).zip](https://github.com/user-attachments/files/19972933/log.30.04.25.zip)

if a simple cube still crash as well?

Jameshigson-PD, 2025-04-30:
I've tried with the primative shapes. The basic shapes all sliced ok but when I tried the Benchy it crashed. I turned on tree support, sliced, clicked send, re-syned the printer info and then it crashed. Please see log file attached.

[log (30.04.25 - 2).zip](https://github.com/user-attachments/files/19976185/log.30.04.25.-.2.zip)

Jameshigson-PD, 2025-04-30:
I've tried again with the benchy and no support, it crashed after slicing, before I could click print
my click path: Open BBS, Sync printer info, continue to sync filament, Overwrite, ok, right click model, change filament, select PLA matte, select 0.16 balanced quality, slice, CRASH. 

Not sure what other combinations to try.

Jameshigson-PD, 2025-04-30:
Open BBS, prepare, Add primative cube, sync printer info, continue to sync filament, overwrite, ok, slice in filament saving mode, print plate, send, successfully sent. 

Not sure why it works with just primative shapes, not a benchy within the software or any of my own files.

Jameshigson-PD, 2025-05-02:
@Haidiye00 any updates please?

OpenWorlds1998, 2025-05-07:
Hey, I have the same issue - it will crash on most large or complicated prints. I'm only using 16 of my 32GB of RAM, but my CPU is nearly maxed out. I did read somewhere that the 13th generation intel processors were reportedly giving some people issues. It was a scenario of the CPU getting nearly maxed out to where it ran too hot, using more power than it should - taking that power away from the GPU (in my case it's a NVidia T400) and causing it to crash. Dunno if this is the problem or not. BUT I too, use mine at work and if I can't slice prints, it slows down our work. Please fix this as it is a reoccurring issue for me.

Haidiye00, 2025-05-08:
Hello, I'm very sorry. Could you please find the log folder in the following path, delete this folder, and then open it with the latest version of the zip file to reproduce your crash operation, and then re upload this log folder? Thank you~This will make it easier for us to locate and analyze the problem. It would be even better if reproducible GIFs and related 3mf files could be provided
[image]
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

Jameshigson-PD, 2025-05-08:
[image]

OpenWorlds1998, 2025-05-08:
Good morning. Yes, here are the files you requested: 

[log.zip](https://github.com/user-attachments/files/20101446/log.zip)

Thank you for checking it out!

Jameshigson-PD, 2025-05-08:
I've had the software successfully send simple primatives to print. The more complex the file the more likely it is to crash, it just failed again trying to print a large more detailed model.

OpenWorlds1998, 2025-05-08:
Here is the video - for some reason I can't upload GIFs:

[record_2025-05-08_07-05-45.zip](https://github.com/user-attachments/files/20101880/record_2025-05-08_07-05-45.zip)

Haidiye00, 2025-05-08:
[image]
Hello, could you please upload the file from this location again? Thank you

Jameshigson-PD, 2025-05-08:
[NTDLL FILE.zip](https://github.com/user-attachments/files/20103538/NTDLL.FILE.zip)

Jameshigson-PD, 2025-05-14:
Update - still having this issue

Jameshigson-PD, 2025-05-14:
Issue is clearly isolated to sending from windows to H2D. 

The issue does not occur when sending from windows to X1C, or from mac software to H2D. 

Please, fix this asap, it's very very frustrating.

OpenWorlds1998, 2025-05-15:
Hey, for me it's been resolved - I updated my graphics drivers again and now it's fixed. Works - so-far so-good.

Jameshigson-PD, 2025-05-15:
@OpenWorlds1998 Glad it's stopped happening for you. I also updated my drivers and even the system bios. The issue went away for a few days and now it's come back. If it starts happening again post again please.

Jameshigson-PD, 2025-05-20:
@Haidiye00 any updates?

Jameshigson-PD, 2025-05-22:
Still experiencing the same issue H2D crashes every time

OpenWorlds1998, 2025-05-23:
Have you tried setting your cpu affinity to only use, say 2 or 4 cores, for bambu studio?
ps - it will take a lot longer to slice

Haidiye00, 2025-06-03:
Hello, please use the latest version of the software. Thank you~
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59

[image]
If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

closed 2025-06-03 by Haidiye00