bambulab/BambuStudio #6677: Logged out of Bambu Studio every time I change location

rboisjoly, 2025-04-24:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS X 15.3.1

### Printer

X1C and A1

### How to reproduce

Stay logged into Bambu Studio, put Mac to sleep, wake up in another location on another network

### Actual results

Cannot see either printer, even the local one. Message that Failed to connect to the server
Then, when logging in, must enter full email and password, etc. Does not work with Passwords pon Mac OS which is also annoying.

### Expected results

To be logged back in automatically
Alternatively, if logged out, to actually be logged out which is not the case as I have to Log out and log back in and enter all credentials manually. I would hope to at least be able to enter password from Password app.

### Project file & Debug log uploads

Does not apply

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-24:
Thank you for your feedback. Could you provide us with the screen recording and log files for our analysis?

rboisjoly, 2025-04-24:
> BoBoYu-BamBu

Log files for Bambu Studio?

Renaud Boisjoly
Founder & CPO
***@***.***
Toll free: 1 (844) 877-3116
Direct: 1 (514) 296-3971

www.studyo.co

￼

rboisjoly, 2025-04-24:
A screen recording is difficult, as this happens when I move from one place to another which doesn’t happen every day and takes a couple of hours… I cannot record when the computer lid is closed

Renaud Boisjoly
Founder & CPO
***@***.***
Toll free: 1 (844) 877-3116
Direct: 1 (514) 296-3971

www.studyo.co

￼

BoBoYu-BamBu, 2025-04-25:
@rboisjoly As shown in the figure below, click on the drop-down box of the file, select "Help", choose "Show Configuration Folder" to open it, find the log file, and then send it to me for all the contents to be packed.
Image
Image

rboisjoly, 2025-04-25:
Ok, here you go

Could not find a Log file with a mod date specifically when I experienced the issue last, but maybe within the log files I am providing you might be able to see something

￼