bambulab/BambuStudio #6679: Bambu Studio will not open file explorer

benkooienga, 2025-04-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2 Build 22631.5189

### Printer

X1 Carbon

### How to reproduce

1. Slice model
2. Export plate sliced file
3. Save file to desired location
4. Popup appears saying "Export Successfully" with link to Open Folder
5. Click "Open Folder"
6. Nothing happens

### Actual results

Clicking "Open Folder" does not open anything

I would like to also point out that the "Show configuration folder" option also does not work.

### Expected results

1. Slice model
2. Export plate sliced file
3. Save file to desired location
4. Popup appears saying "Export Successfully" with link to Open Folder
5. Click "Open Folder"
6. Windows explorer opens to the correct location of the sliced file.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/19893633/Cube.zip)

[debug_Wed_Apr_23_08_36_23_40968.log.zip](https://github.com/user-attachments/files/19893672/debug_Wed_Apr_23_08_36_23_40968.log.zip)
[log_iotc.txt](https://github.com/user-attachments/files/19893673/log_iotc.txt)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-25:
@benkooienga  Thank you for your feedback. Could you please provide a screen recording? It would be helpful for us to analyze and reproduce the problem.

benkooienga, 2025-04-25:
> [@benkooienga](https://github.com/benkooienga) Thank you for your feedback. Could you please provide a screen recording? It would be helpful for us to analyze and reproduce the problem.

I've attached a zip file of the screen recording. At the end I am clicking the "Open Folder" link, but nothing is happening.

[Screen Recording 2025-04-25 091146.zip](https://github.com/user-attachments/files/19909239/Screen.Recording.2025-04-25.091146.zip)