bambulab/BambuStudio #6683: Program Crashing When Trying to Slice Plate

dirkviester, 2025-04-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 9 7940HS
Memory: 32 GB
Display Card: Radeon 780M

### Printer

Bambu X1-Carbon

### How to reproduce

1. Have a object on the plate.
2. Click "Slice Plate"

### Actual results

Program starts slicing, but crashes before completing.

### Expected results

Program completes the slice and the object is ready to be sent to the printer.

### Project file & Debug log uploads

[Assemblyzzz.zip](https://github.com/user-attachments/files/19896696/Assemblyzzz.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-04-25:
Thanks for your feedback and I have tried your 3mf and it seems no crash happened.
[image]

Can you do:
- Upload your operation gif to show that?
- Upload your crash dmp file(I notice you post crash log in zip and it is not enough) for us to analyze. You can follow https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue.

QingZhangBambu, 2025-04-29:
@dirkviester Could you try it to see if it works
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

JohnSmithCYM, 2025-05-27:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657/artifacts/3189644769
Hello, @Haidiye00 have compiled a temporary package here. Can you download it and help me see if it can slice without crashing~thanks