bambulab/BambuStudio #6685: Slice never finishes, hang up studio until force quit

nochotwo, 2025-04-24:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

A1. P1S

### How to reproduce

load model
slice
hang on slice

### Actual results

[image]

### Expected results

slice should complete

### Project file & Debug log uploads

[hamster_split.zip](https://github.com/user-attachments/files/19897078/hamster_split.zip)

[log.zip](https://github.com/user-attachments/files/19897085/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-04-25:
[image]

Thanks for your feedback, we have found it is a problem, and you can trun off 'Detect floating vertical shells' to avoid that.

XunZhangBambu, 2025-06-03:
@nochotwo Hi,this has been fixed in our latest version 02.01.00.59