bambulab/BambuStudio #6687: Stopped recognizing SD card

lobert, 2025-04-24:
### Bambu Studio Version

2.0.0.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Printer

Bambu Lab P1P

### How to reproduce

Does not recognize SD card (mobile app does, however).

### Actual results

Does not recognize SD card (mobile app does, however).

### Expected results

SD card should be available

### Project file & Debug log uploads

[debug_Thu_Apr_24_15_18_34_23954.log.0.zip](https://github.com/user-attachments/files/19900573/debug_Thu_Apr_24_15_18_34_23954.log.0.zip)

[Chair cylinder stop.3mf.zip](https://github.com/user-attachments/files/19900576/Chair.cylinder.stop.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-25:
@lobert  Thank you for your feedback. Could you please take a screenshot and show it to us? We can't figure out what error occurred when we couldn't recognize the SD card.

lobert, 2025-04-25:
> BoBoYu-BamBu

￼

lobert, 2025-04-25:
By the way, I see on the forum that others are complaining that the two recent updates are unable to recognize the card.

<Screen Shot 2025-04-24 at 10.14.44 PM.jpg>

mybabysexy, 2025-04-25:
Same, had to revert to `Bambu_Studio_win_public-v01.10.02.76-20250225202310`  in order to send the file to the printer (A1 in LAN mode)

SonOfAbaddon, 2025-04-25:
Hi, we already reported similar issues here: https://github.com/bambulab/BambuStudio/issues/6235
So far no solution with all 2.XX releases.