bambulab/BambuStudio #6690: Camera stream not working on H2D

AnalogKnight, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu lab H2D

### How to reproduce

1. Check the “Go Live” box
[image]
2. Config Media Source in OBS Studio
[image]

### Actual results

Nothing appears in OBS Studio, but it works fine on my X1C.

### Expected results

The printer camera should be displayed in OBS.

### Project file & Debug log uploads

This operation does not seem to generate any Logs, please let me know if I am mistaken.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-25:
Sorry, currently GO Live does not support Sangtek. The development is scheduled to be carried out and we are looking forward to your coming. Please be patient.

bambu123, 2025-04-25:
Go Live does not support Agora network used by H2D yet.
Please wait.

JonLaliberte, 2025-05-30:
I can't seem to stream from H2D to Home Assistant either. 
For example with a URL like: `rtsps://bblp:XXXXXX@YYYYYYY:322/streaming/live/1 `