bambulab/BambuStudio #6692: Login window is blank

ulbin, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

CPU: AMD Ryzen 7 9700X 8-Core Processor                3.80 GHz
RAM 64 GB
GPU: Nvidia RTX 4070 Super

### Printer

Bambu Lab X1C

### How to reproduce

At login in Bambu studio white window is displayed, but it remain blank and iti is impossible to login.

### Actual results

[image]

### Expected results

I would be able to enter username and password and login.

### Project file & Debug log uploads

No specific project.

[Bird+Self-Watering+Spike+2.zip](https://github.com/user-attachments/files/19904489/Bird%2BSelf-Watering%2BSpike%2B2.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-25:
Sorry, could you try other networks? Also, please provide the log.

ulbin, 2025-04-25:
Log of what? I just can not login as windows is displayed showing nothing.

 

I am on network and login in makerworld works fine, but I can not login in Studio and can not connect or print over the cloud.

BoBoYu-BamBu, 2025-04-25:
@ulbin  As shown in the figure below, click on the drop-down box of the file, select "Help", choose "Show Configuration Folder" to open it, find the log file, and then send it to me for all the contents to be packed.
[image]
[image]

ulbin, 2025-04-25:
image.png (view on web) <https://github.com/user-attachments/assets/5e46f17d-27c0-4a34-a7b3-d02c6d9ca090> 
image.png (view on web) <https://github.com/user-attachments/assets/767fccbb-9360-4711-8a47-b3b268240848> 

[25-04-25 09:58:04.076716][D][3F8C][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-25 09:58:04.077716][D][3F8C][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-25 09:58:04.077716][D][3F8C][IOTC] @ErrCode -12 - Line (13010)
[25-04-25 09:58:34.846671][D][9658][IOTC] [IOTC_Check_Session_Status] Error: Not Initialized!
[25-04-25 09:58:34.846671][D][9658][IOTC] @@@@@ IOTC_DeInitialize called
[25-04-25 09:58:34.846671][D][9658][IOTC] @ErrCode -12 - Line (13010)

ulbin, 2025-04-25:
Hi,

 

I have found solution https://www.reddit.com/r/BambuLab/comments/154db9i/bambu_studio_login_issue/ and it works now. 

 

But I do not know how the language field was erased as I had no such problem with previous versions of the program.

 

Regards,

 

Miran Ulbin