bambulab/BambuStudio #6693: file rejected do the printer same file runs by the bambu app

spigh12, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Version 22H2 voor x64 (KB5057056)

### Printer

lab x1 carbon

### How to reproduce

printer rejects file transport from Studio     0500-4003-133323

### Actual results

0500-4003-133323

### Expected results

totaly no solotion if i try again

### Project file & Debug log uploads

any file does not work

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-25:
Sorry, could you record a screen to describe your problem for me?

BoBoYu-BamBu, 2025-04-27:
@spigh12 Hello, could you please provide the log file and the screen recording? Thank you.

spigh12, 2025-04-27:
After i reisall an older version i became a problem from a failing network isseuI deleted the hole Bambu dect from my labtop and did a disk cleanupThen i install the newer version + the extenion Then everything workt as it should beIm don t understand what better this new version can be because it all workt fine for meThe newer Look on the x1 carbon is not neceserialy an the new Slogans only work on my NervesVriendelijke groet Albert van Zanten

closed 2025-04-29 by BoBoYu-BamBu