bambulab/BambuStudio #6694: Exhaust Fan Control NOT working in new BETA firmware or Bambu Studio

guywholikestobreakthings, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

11 win

### Additional system information

As the title says, just got forced into a Beta update (well, i clicked update) and now I can't change the exhaust fan speed. Get an error, then can turn it on/off only on the printer. When you click +/- speed, it doesn't react, and the Bambu Software doesn't allow it either. It says it's only available for PETG/PLA/TPU but i'm printing PET. This is the first time i've ever seen this one.... I used to change the exhaust fan speed all the time as i've got a bento and don't want to listen to a jet engine in my office.

### Printer

X1c

### How to reproduce

As the title says, just got forced into a Beta update (well, i clicked update) and now I can't change the exhaust fan speed. Get an error, then can turn it on/off only on the printer. When you click +/- speed, it doesn't react, and the Bambu Software doesn't allow it either. It says it's only available for PETG/PLA/TPU but i'm printing PET. This is the first time i've ever seen this one.... I used to change the exhaust fan speed all the time as i've got a bento and don't want to listen to a jet engine in my office.

### Actual results

As the title says, just got forced into a Beta update (well, i clicked update) and now I can't change the exhaust fan speed. Get an error, then can turn it on/off only on the printer. When you click +/- speed, it doesn't react, and the Bambu Software doesn't allow it either. It says it's only available for PETG/PLA/TPU but i'm printing PET. This is the first time i've ever seen this one.... I used to change the exhaust fan speed all the time as i've got a bento and don't want to listen to a jet engine in my office.

### Expected results

As the title says, just got forced into a Beta update (well, i clicked update) and now I can't change the exhaust fan speed. Get an error, then can turn it on/off only on the printer. When you click +/- speed, it doesn't react, and the Bambu Software doesn't allow it either. It says it's only available for PETG/PLA/TPU but i'm printing PET. This is the first time i've ever seen this one.... I used to change the exhaust fan speed all the time as i've got a bento and don't want to listen to a jet engine in my office.

### Project file & Debug log uploads

As the title says, just got forced into a Beta update (well, i clicked update) and now I can't change the exhaust fan speed. Get an error, then can turn it on/off only on the printer. When you click +/- speed, it doesn't react, and the Bambu Software doesn't allow it either. It says it's only available for PETG/PLA/TPU but i'm printing PET. This is the first time i've ever seen this one.... I used to change the exhaust fan speed all the time as i've got a bento and don't want to listen to a jet engine in my office.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-27:
@guywholikestobreakthings Sorry, could you please provide the screenshot of the error and the log file?

guywholikestobreakthings, 2025-04-27:
https://drive.google.com/file/d/1R22qDK2y7lGfXnHCmQIchaFNd14PcAqo/view?usp=drivesdk

closed 2025-05-29 by BoBoYu-BamBu