bambulab/BambuStudio #6695: The save file operation failed to connect to the open and save panel service.

ziem, 2025-04-25:
### Bambu Studio Version

2.0.1.50, 2.0.3

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Montenery 12.7.6

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Generate GRIPS plate (with Smallest build plate dimension (mm) set to 170) https://gridfinity.perplexinglabs.com/pr/grips/0/0
2. Download it
3. Open Bambu Studio
4. Create new project
5. Drag and drop downloaded STL file
6. Select the plate and select split -> to objects
7. Then press CMD + A to select all
8. Select Arrange all objects -> Arrange
9. Select the first plate
10. Press Slice plate
11. Press Export plate sliced file
12. 💥 `The save file operation failed to connect to the open and save panel service.`

This happens on freshly installed Bambu Studio with Europe region selected and Network Plugin not installed.

### Actual results

Error with the following message:
```
The save file operation failed.
The save file operation failed to connect to the open and save panel service.
```

### Expected results

No error, and file correctly exported.

### Project file & Debug log uploads

[6b046c2bd265cb80900bcadd2fc1e21cd11939e1.3mf.zip](https://github.com/user-attachments/files/19908601/6b046c2bd265cb80900bcadd2fc1e21cd11939e1.3mf.zip)

[debug_Fri_Apr_25_14_29_30_90820.log](https://github.com/user-attachments/files/19909211/debug_Fri_Apr_25_14_29_30_90820.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-27:
@ziem Thank you for your feedback. Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

Haidiye00, 2025-04-27:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

ziem, 2025-04-27:
Here is the recording:
[image]

jdenavascues, 2025-09-03:
I have the same problem - I checked that BambuStudio had disk permissions, but I still get the same error in MacOS Monterey. Is there any way to address this issue? I do not have a more modern computer I can rely on.

jdenavascues, 2025-11-15:
Hi,
In my case, the problem was that I was using an old Mac and the MacOS version was too old. I can work perfectly fine with a more modern computer/OS.

Haidiye00, 2025-11-16:
Hello, this macOS Montenery 12.7.6  version is indeed too low. Currently, the lowest version in our laboratory is mac13. We suggest upgrading to the Mac system version