bambulab/BambuStudio #6698: Wall anchors below bridge layer are missing on internal geometry

gary-wlg, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4.1

### Printer

All

### How to reproduce

The 2 layers prior to bridging over sparse infill is suppose to generate an internal solid infill "perimeter / anchor" around the walls so that the bridge layer above is properly supported along the edges. 

Starting sometime with versions 2.0.x only outer walls had this solid infill "perimeter / anchor" generated.

### Actual results

Missing sparse infill anchor perimeter below the bridging layer is missing around internal geometries.  In previous version 1 slicer, the bridging anchor perimeter went around all perimeters so that the bridge layer is properly supported.

attached are images from both slicer versions.

[image]

[image]

### Expected results

I would expect there to be a bridging anchor / perimeter around all perimeters so that the bridge layer has something to anchor to. 

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gary-wlg, 2025-04-25:
I have confirmed the issue is not present in version 1.10.2.76, but it is an issue with all versions after this.

XunZhangBambu, 2025-05-14:
@gary-wlg  Could you provide the 3mf to help us analysize. I cannot reproduce in my model

[image]

ddremann, 2025-05-15:
I have been struggling with this issue as well.  This effects the print quality on many of my designs as the bridging layer will pull back from the wall due to no sparse infill anchor perimeter below the bridging layer.  Thanks @gary-wlg for the workaround as I can confirm that version 1.10.2.76 does not have this issue.

v02.00.03.54
[image]

v01.10.02.76
[image]

ddremann, 2025-05-16:
Here are some test sample 3mf files (renamed to .txt) that show the issue.  I also noticed that the issue occurs when using the standard 0.20mm preset but only when changing the [Top shell layers] to 3 or less.  Let me know if this helps.

[v01_10_02_76.txt](https://github.com/user-attachments/files/20235901/v01_10_02_76.txt)

[v01_10_02_76.gcode.txt](https://github.com/user-attachments/files/20235902/v01_10_02_76.gcode.txt)

[v02.00.03.54.txt](https://github.com/user-attachments/files/20235903/v02.00.03.54.txt)

[v02.00.03.54.gcode.txt](https://github.com/user-attachments/files/20235904/v02.00.03.54.gcode.txt)

XunZhangBambu, 2025-05-16:
@ddremann  Thanks for your 3mf. We have found the root cause, will fix it soon

gary-wlg, 2025-05-16:
Thanks @ddremann for providing the sample 3mf's for this issue, I appreciate it.

Excellent new the root case has been found, can't wait for the fix.