bambulab/BambuStudio #6701: Error -5010

Bierbantscher, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Slice the Model
2. Send to Printer
3. The process stops at 99%
4. Error -5010, the G-Code couldn´t be send
5. 

[image]

### Actual results

The G-Code is on the SD-Card and can be printed

### Expected results

There´s an Error Message but the process works correct

### Project file & Debug log uploads

[Körper23.zip](https://github.com/user-attachments/files/19914976/Korper23.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Bierbantscher, 2025-04-26:
I updated to Firmware 01.08.50.20 Beta today, till the Update the Printer was running on Firmware 01.08.50.18 Beta

If i press send, the Upload to the Printer always stops at 99%
[image]

After a few Moments i get these error Message
[image]

The .3mf File is on the SD-Card
[image]

Sometimes it´s possible to print the File, sometimes not

I tested it with Random Files, it´s always the same result.

If i choose the Option Print plate, there´s no problem

BoBoYu-BamBu, 2025-04-27:
@Bierbantscher  Thank you for your feedback. Could you try again with a different network? If it still doesn't work, please provide the log file for us to analyze the problem.

BoBoYu-BamBu, 2025-04-29:
@Bierbantscher Hello, have your questions been resolved?

closed 2025-05-22 by BoBoYu-BamBu