bambulab/BambuStudio #6703: Login/Register is a blank page

Muzyzy, 2025-04-25:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home Version 2009 

### Additional system information

Device Name	DESKTOP-IO83JJS
Processor	Intel(R) Core(TM) i5-10400F CPU @ 2.90GHz   2.90 GHz
Installed RAM	16.0 GB
Storage	477 GB SSD TEAM T253X2512G
Graphics Card	Radeon RX 5500 XT (4 GB)
Device ID	5E221929-A151-48BB-A91F-B123E2A2584C
Product ID	00325-81426-46026-AAOEM
System Type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 10 Home
Version	2009
Installed on	‎29/‎7/‎2567 (Thai Calender) equiv. 29/7/2024
OS Build	19045.5737

### Printer

P1S

### How to reproduce

1. Press Login/Register
2. Window pop up, but blank forever

### Actual results

[image]

### Expected results

prompt login details?

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19917332/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

MartinChan3, 2025-04-26:
Can you change your language to other language and give it a try again? And what's more, which language are you using?

Muzyzy, 2025-04-26:
Hi Martin, I have changed the calendar to english and now it displays 2025, however problem still the same. I have always use english for my pc

JohnSmithCYM, 2025-04-27:
Can you offer your log here? Thanks again.

Muzyzy, 2025-04-27:
> Can you offer your log here? Thanks again.

here you go
[log.zip](https://github.com/user-attachments/files/19928948/log.zip)

Muzyzy, 2025-04-27:
> Can you change your language to other language and give it a try again? And what's more, which language are you using?

Hi, It worked. I changed the language in Bambu Studio. Preference > Language, and It restarted the app, now I am able to log in. Thank you.

realzorro, 2025-04-28:
You can try the following method: change the language to another language in the configuration interface, and then switch back to English to see if it can solve your problem