bambulab/BambuStudio #6704: Filament sync screen does not show external roll when choosing print from bambu studio print menu. sync screen does not appear when sync icon is pressed but dialog box says filaments have been synced

rajiam, 2025-04-25:
### Bambu Studio Version

mac-v02.00.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

Mac 12.7.6

### Printer

bambu x1

### How to reproduce

1. turn on printer; 2. choose control screen; 3. choose filament screen; 4. load Generic TPU filament into printer until old filament is purged; 5. open bambu studio; 6. choose sync filament icon; 7. sync filament dialog shows external roll unmapped.; 8. I choose roll 17 "unmapped" and I choose EXT roll icon from dropdown menu; 9. roll 17 icon changes to 17 TPU EXT; 10. I press synchronize now and dialog closes. 11. I pick object and choose external filament 17 and the object changes to that color on the build plate; 12. I preview plate; 13. I choose Print Plate and the print dialog comes up but the EXT slot is unmapped. 14. The TPU filament swows up but has a red exclamation point; 15. I pick the unmapped TPU filament and a box drops down but is missing the EXT slot or else it is not showing because the bottom of the dialog is not showing and there is no scroll bar to get to it and choose it. 

So the mapping in the sync dialog is buggy. and the mapping in the print dialog is also buggy. This is a problem in this release and the last one. everything worked before the last 2 releases.

[image]

[image]

[image]

[image]

[image]

### Actual results

The sync dialog does not properly map or sync the EXT filament slot.
The print dialog box does not "KNOW" that the filaments have been synced and shows EXT slot unmatched, and does not allow mapping to the EXT slot. See 5 screenshots above.

### Expected results

1. I expect the filaments to sync when I sync them and stay synced after the program reports that the sync was successful.
2. I expect to be able to pick any slot from the print dialog box. I expect all slots to be accessible to be picked from the print dialog so I can use my external slot.

### Project file & Debug log uploads

[rabbitbenchySUPERDUAL_merged.gcode.3mf.zip](https://github.com/user-attachments/files/19917380/rabbitbenchySUPERDUAL_merged.gcode.3mf.zip)

the only log files I can find are in my config folder. there is no Bambu Studio/log directory in my Library/Application support directory. I tried to drag into this box but they are wrong file type.

### Checklist of files to include

- [ ] Log file
- [x] Project file

rajiam, 2025-04-25:
Tried in Bambu Handy. Everything works fine in Bambu Handy. It must be a bug in the last 2 versions of Bambu Studio!!

rajiam, 2025-04-26:
I was able to work around this bug by not syncing. I replaced generic TPU with Geeetech TPU in Bambu Studio filament list. then sliced and exported to sd card. I inserted sd card and chose the file to print, Then unchecked AMS and was able to pick TPU which was not mapped.

XinZhangBBL, 2025-05-06:
@rajiam Due to the screen size, a part of it is not displayed. You can drag the print sending window upwards on the screen, and then click to perform the mapping. 

[image]

rajiam, 2025-05-06:
This does not work!!!  My bambu program window is full size, the print window is at the top of the program window and the dropdown filament window in the print window is at the very bottom of the computer screen and has no scroll bar. The bottom of the AMS 4 is as low as the window will show and does not even show AMS 4 label. there is no scroll bar to get below the AMS 4. It is impossible to pick if it is not listed. Bug.

[image]

As you can see the Dropdown menu is at the very bottom of the screen and does not even show the bottom of AMS4, and has no scroll bar. this is a Bambu Studio bug and should be fixed ASAP. The TPU roll in the EXT slot is grey but the resic window lets me change and confirm to grey, but when I open the re-sync button immediately after, it shows the EXT slot unmatched. So I cannot properly sync my EXT slot from the re-sync window. That is why the EXT slot shows unmatched EXT slot. then I cannot change it from the print window either. the only way I can print with the EXT slot is to make the AMS off choice in the print window. So there is a bug in the sync window and the print window because it just does not work properly for the EXT slot when there are 4 AMS units because there is no space to display the EXT slot below the 4th AMS unit in the filament dropdown menu in both the sync window and the print window. Bug. Bug.  Bug.

rajiam, 2025-05-06:
I can change the EXT slot from 'unmatched' to grey in the sync window by double clicking it. I then press resync and the window closes and it confirms. but when I open the window right after I have done this, the EXT slot has reverted to 'unmatched'.

[bambu 2.pdf](https://github.com/user-attachments/files/20052436/bambu.2.pdf)

rajiam, 2025-05-06:
I sill don't get why there are now 2 choices in the sync window. If I change a color and "overwrite" Can I resync to a different color from the slice window again?. Do I have to re open the program to re-sync the colors I overwrote? This is very confusing. The sync window is no longer intuitive to me. I have read over it a few times and it seems redundant, confusing and over complicated. When am I supposed to choose mapping? When am I supposed to use overwriting? Can I remap my filaments after they are overwritten?

rajiam, 2025-05-06:
The print sending window is at the top of the screen. The AMS drop down window touches the bottom of the screen showing only the 4 AMS UNITS and the title of AMS 4 is cut off at the bottom of the screen with no scroll bar on the AMS drop down window. There is no way to choose my EXT SLOT. BUG. BUG. BUG. DID you see my screen shot?  The EXT slot is not available to choose on my portable Mac computer because the choice window is already as far up as possible but the bottom does not show and there is no way to scroll to the bottom of the AMS drop window. BUG. Rajiam

image.png (view on web)

XinZhangBBL, 2025-05-07:
OK, I understand your trouble.  Thank you for your feedback. We will optimize this interaction experience. And we will provide you with a temporary solution: try adjusting the resolution and the font magnification ratio so that the window can be fully displayed.  

> This does not work!!! My bambu program window is full size, the print window is at the top of the program window and the dropdown filament window in the print window is at the very bottom of the [...]

rajiam, 2025-05-08:
I adjusted my screen resolution to 1680 x 1050 and this is a good workaround. I can now see the complete list of filament with the print window moved all the way up. I would still like to use a lower resolution that is easy to read on my 13" MacBook pro. So please implement a scroll bar at your earliest convenience. The Print window make more sense now. I first set the correct filaments in the filament dropdown for each filament to match the filaments in the AMS units in the prepare mode then run the sync window and override and all the filaments correctly match and stay matched after sync.