bambulab/BambuStudio #6705: Cannot calibrate VFR on H2D

psiberfunk, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

H2D

### How to reproduce

Cannot calibrate max VFR for a filament for the H2D printer because there's no Orcaslicer support _(**thanks for breaking that on purposeBambu !**)_  AND you don't show the calibration menu (not tab, but MENU, where you can do the useful non-automatic tests) for your own printers.  H2D printer is the first printer that comes with fully locked-out support for Orcaslicer and 3rd party slicers, so the ONLY way is to use Bambu Studio, which lacks this feature, which IS available for non-bambu printers , ironically.

Please fix this by making the "advanced calibrations" menu available for people who want/need it.

### Actual results

Bambu forgot to think about their users again, see above.

### Expected results

Bambu lets me calibrate max VFR using a standard test.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

psiberfunk, 2025-04-26:
Note:  There's a super-easy fix for this here:

[image]

BUT I can't use it because, again, there's silly (stupid) signing restrictions to actually interact with the printer.   It's unreal to me that you've locked us out of useful fixes we can do ourselves for "security" hand-waving reasons.  This is incredibly frustrating

Qjeezy, 2025-04-26:
Cmon Bambu! We need these calibrations! Sometimes flow rate and dynamics is not enough. Some of us are perfectionists and want to calibrate all aspects of a filament. 

At the very least unlock this menu. Or figure out how to incorporate these calibrations into your own calibration tab right next to flow rate and flow dynamics.

JLRGURU, 2025-04-26:
YES!!! WE NEED IT!!! CMON BAMBU!!!!

psiberfunk, 2025-05-03:
So , uh , is this a bug or not ? It certainly is painful to not be able to do ….

Qjeezy, 2025-05-03:
> So , uh , is this a bug or not ? It certainly is painful to not be able to do …. 

It's not a bug, it's just that Bambu isn't making the in depth calibrations available to us and orca slicer doesn't support the H2D. We're kind of stuck making our own calibrations via custom gcode.

psiberfunk, 2025-05-03:
> > So , uh , is this a bug or not ? It certainly is painful to not be able to do ….
> 
> It's not a bug IMO, it's just that Bambu isn't making the in depth calibrations available to us and orca slicer doesn't support the H2D. We're kind of stuck making our own calibrations via custom gcode.
> 
> Edit: just realized you're OP and what you were referencing.
> 
> Cmon Bambu!!!! We need more calibrations!! Give them to us! They're already there! Just unlock them!

They marked it as a bug, then un-marked it, then re-marked it, then un-marked it again.  It seems they can't decide if making the experience better for other printers **in their own slicer** is a bug, LOL.

Qjeezy, 2025-05-03:
> They marked it as a bug, then un-marked it, then re-marked it, then un-marked it again.  It seems they can't decide if making the experience better for other printers **in their own slicer** is a [...]

Yea I saw that after I hit add comment lol. I agree, they're being thee bug on this one haha.

psiberfunk, 2025-05-14:
@zhimin-zeng-bambulab  and @BoBoYu-BamBu , this really should be considered a bug, considering _**your own marketing copy**_ shows the printer doing VFR flow rate calibrations... 

[image]

p-c-mo, 2025-05-15:
Come on bambu...

psiberfunk, 2025-06-02:
Any time now would be great :)

psiberfunk, 2025-07-02:
Any love for this Bambu ?  It would be nice to just have this really simple thing enabled since I **cant print to your printer with orcaslicer anymore... **

psiberfunk, 2025-08-16:
Still not available in the release.. very disappointing..