bambulab/BambuStudio #6706: Can't print to my A1

POCONOPRINTING, 2025-04-26:
### Bambu Studio Version

2.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel I9 Processor

### Printer

A1

### How to reproduce

I can't select a filament for the A1. All my other printers are working fine. The A1 was fine until the update. 

### Actual results

Wont print

### Expected results

Nothing fixes the issue

### Project file & Debug log uploads

I tried multiple files and nothing works

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2025-04-26:
Can you give a screenshot or video to show the problem? The selection operation happened on 'Send Print Job' or other place? You are trying to print with AMS or ext spool? And is there anything correct in device page?

BoBoYu-BamBu, 2025-04-29:
@POCONOPRINTING Hello，Could you record a screen or take a screenshot to clearly describe your problem?

XinZhangBBL, 2025-06-17:
The problem will be closed. You can reopen it if you meet same problem.

closed 2025-06-17 by XinZhangBBL