bambulab/BambuStudio #6708: when using inner/outer/inner print order the speed profile used also swapped causing the outer wall to print with inner wall speed

boysie123, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

Bambu lab p1p

### How to reproduce

with a 2 wall print
using any print set the inner wall to be fast and the outer wall to be slow speed so you can visually see the difference or check the gcode
Select print order to inner wall/outer wall/inner wall
User Arcane wall generator not sure this matters but i had it set so listed it here

### Actual results

Outer wall takes on speed of internal wall speed, it might also happen on inner/outer order but check as i expect so. 

### Expected results

Outer wall always takes outer wall speed and not inner wall speed.

### Project file & Debug log uploads

Any project will do something simple its easy to see the issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-04-27:
OK， i will check it.

QingZhangBambu, 2025-05-09:
@boysie123 Could you provide 3MF files? I can not reproduce it.

[image]