bambulab/BambuStudio #6709: crashing axes after new fw popup request

Winky1st, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 

### Printer

Bambu Lab P1S

### How to reproduce

I was re printing from internal memory a file I've printed many times.
My actual fw is 01.07.50.18
Wile printing, on the printer display, it popped up a requester informing me that a new FW version is available giving the possibility to update or cancel. NOTE : I did not any action as I was printing,  waiting for the end of print to get any further action.
But the printer lost his position, the head went full right side and full front side starting to "brrrrrr brrrrr brrrrr " as she came to the end of the line and pushing with motors still. 
After few second of this hugely noise she get back - without any action by me - on the perfect print position and she was able to end the print correctly. (no step misalignments)  
I am sure is not related to Bambu Studio, and I do apologize if it is not the correct place for this kind of report but I couldn't find a different, more specific fw related one
wish you a great day - cheers

### Actual results

at the end the print was anyhow perfectly

### Expected results

Donno what to write here lol - I do not think it is something related to Babu Studio - I guess wile printing, the above mentioned popup, should not come until the end of print or at least not interact with the actual print.
cheers

### Project file & Debug log uploads

[Backpacking whistle.zip](https://github.com/user-attachments/files/19922066/Backpacking.whistle.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

gaolegao-lx, 2025-04-28:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Winky1st, 2025-04-28:
Ok thankyou, done! 
Ticket n: US250428510002

cheers

closed 2025-05-29 by BoBoYu-BamBu