bambulab/BambuStudio #6711: Selection of filament for calibation

bottle12, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.4.1

### Printer

X1C

### How to reproduce

I have two ams 
Go to calibration and  flow dynamics
click A1 
you will see that it autoselects  B1 as wel
Deselect B1 and it also deselects A1

### Actual results

Unable to calibrate

### Expected results

Selected A1 Filament is selected and not B1
This is the same for the other filaments

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-27:
Thank you for your feedback. Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

bottle12, 2025-04-27:
https://github.com/user-attachments/assets/11287334-17e5-4804-aadc-7e03697a2dd3

here is the clip to reproduce