bambulab/BambuStudio #6713: Bambu Studio freeze plate page

matteocaretta72, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel core I5 ultra 13th
16 Gb ram
Intel Graphics

### Printer

Bambu Lab X1C

### How to reproduce

1. Apertura Progetto
2. Provare a selezionare più oggetti con shift e il trascinamento del mouse
3. A volte anche selezionando singolarmente gli oggetti

### Actual results

La pagina dei piani si blocca come fosse congelata, il resto funziona

### Expected results

Il programma non si deve congelare

### Project file & Debug log uploads

[Pressa Per inserti 2.zip](https://github.com/user-attachments/files/19924350/Pressa.Per.inserti.2.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-04-27:
[image]

Here is my gif of your 3mf and it moves smooth. Can you give us a screenshot to show the 'frozen' plate?

matteocaretta72, 2025-04-27:
when I try to select multiple objects with shift and mouse or ctrl + A the workspace becomes unresponsive

 
20250427105047_rec_.gif (view on web)

JohnSmithCYM, 2025-04-28:
[image]
I have tried your description and it seems no move error happened. Is these anything wrong in my operation?

closed 2025-04-29 by JohnSmithCYM