bambulab/BambuStudio #6715: Bambu Studio V2 / Bambu Handy Load Filament Setting for P1P Missing / Not Working

GEGR667, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

I've just download Bambu Studio V2, and the updated the Bambu Handy app for my smartphone (iPhone 13). 
Load Filament on the Bambu Studio is not working. I can unload filament, but the load filament button is greyed out. Similarly, in Bambu Handy, the Load Filament feature is missing.
I have to manually increase the hotend temperature, and manually extrude the filament in order to load it.
Rather sloppy implementation of the new Bambu Studio / Handy apps for P1P

### Actual results

I have to manually increase the hotend temperature, and manually extrude the filament in order to load it.

### Expected results

There should be a load filament option to begin with. Just like there used to be.

### Project file & Debug log uploads

Nothing to upload.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-27:
@GEGR667  Thank you for your feedback. Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

BoBoYu-BamBu, 2025-05-22:
@GEGR667 Hello, may I ask if your problem has been solved?

closed 2025-05-23 by BoBoYu-BamBu