bambulab/BambuStudio #6718: Flush Multiplier Ignored at 0

timbervolk, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab H2D

### How to reproduce

1. Go to 'Flushing volumes'
2. Set Flushing Multiplier to 0 in both nozzles
3. Recalculate flushing volumes; should all be 0
4. Print object that utilizes at least 2 materials in a single nozzle
5. Observe flushing volume

### Actual results

Printer flushes per normal, despite values being set to 0.

### Expected results

Printer skips flushing and immediately begins printing next color.

### Project file & Debug log uploads

[2-Color Combination, Inverted Frames.zip](https://github.com/user-attachments/files/19925629/2-Color.Combination.Inverted.Frames.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

timbervolk, 2025-05-19:
Any update on this? I'm still experiencing this issue. I also confirmed this is only affecting my H2D; my P1S could not reproduce the issue.

paulohenriquesi, 2025-07-09:
Hi! Same problem here. Last version of BS on Mac and H2D updated

grimgrimms, 2025-08-15:
Sign me up for this one effecting me.  Flushing volumes are ignored on the H2D.

XunZhangBambu, 2025-08-15:
@paulohenriquesi @grimgrimms @timbervolk  In our latest version, we have fixed this problem