bambulab/BambuStudio #6719: crash

Mvk76, 2025-04-26:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro for Workstations

### Additional system information

Intel(R) Xeon(R) W-2225 CPU @ 4.10GHz   4.10
32.0 GB (31.7 GB usable) 
64-bit operating system, x64-based processor

### Printer

bambu lab x1, bambu lab P1s, bambu lab a1 mini. 

### How to reproduce

Crashes and closes after I hit slice button. has been doing this for past five updates. 

### Actual results

crashes and closes

### Expected results

crashes and closes

### Project file & Debug log uploads

[2X2X3 Jar Vase Profile New.zip](https://github.com/user-attachments/files/19925587/2X2X3.Jar.Vase.Profile.New.zip)

[debug_Sat_Apr_26_17_51_04_21888.log.zip](https://github.com/user-attachments/files/19925635/debug_Sat_Apr_26_17_51_04_21888.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-04-27:
[image]
It works well on my pc and your log file is empty(1KB). Can you upload your log file and dmp file(it is needed in your situation) again? https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Mvk76, 2025-04-27:
[log.zip](https://github.com/user-attachments/files/19927949/log.zip) Here it is.

QingZhangBambu, 2025-04-29:
@Mvk76 Could you try it to see if it works
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

JohnSmithCYM, 2025-05-27:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657/artifacts/3189644769
Hello, @Haidiye00 have compiled a temporary package here. Can you download it and help me see if it can slice without crashing~thanks

Mvk76, 2025-05-29:
> [@Mvk76](https://github.com/Mvk76) Could you try it to see if it works https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

Didn't help.  thank you though.