bambulab/BambuStudio #6720: Duplicate copy

frankmitchell4741, 2025-04-27:
### Bambu Studio Version

Latest

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

I consider myself an expert in printing as I have been doing this for 10 plus years and own 7 Bambu labs machines including the new H2D Just giving a little context to validate my experience.
I have an issue with the new Bambu Studio Software.
Randomly when I open a saved print the software will duplicate an exact copy of the part on top of an existing part I try to slice it and it looks normal until I send it over to the machine to print then the Chaos unfolds. the machine tries to print both parts in the same space and it gets an excess amount of material build up. It is so bad that the part looks terrible and I must scrap the part.
This is a daily annoyance as I run my machines every day I am losing money over scrapped parts.
I should not have to check your software each time.
I have seen others having the same issues ever since the new update to the software that included the H2D.
Please fix this!

### Printer

X1C,H2D

### How to reproduce

It's random when I reopen file

### Actual results

It duplicates part exactly over part

### Expected results

I expect it to not duplicate parts without my intentional duplication

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-27:
@frankmitchell4741 Sorry,Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

frankmitchell4741, 2025-04-27:
It's random so hard to catch it when it does it 
I can close file several times and can't duplicate the results then when I open a file it will duplicate a copy right on top of a copy and I usually don't notice until I print it and it fails .
I can say it does not happen when the  file is left open for multiple sessions

joxebike, 2025-05-07:
Same problem here V 20.0.1.50 with W10 22H2
Project with 4 plates, open, move one model a bit save and close. all is ok.
Open again and every plate has 2 copies of the same item in the same position. On plates with multiple models, of them are duplicated
Sources models are .3mf

joxebike, 2025-05-07:
[Patas para reposapies escritorio.zip](https://github.com/user-attachments/files/20092686/Patas.para.reposapies.escritorio.zip)

joxebike, 2025-05-07:
https://github.com/user-attachments/assets/cbf2dada-861f-4169-b93a-659d36832b91

frankmitchell4741, 2025-05-07:
I still have no solution lost a print earlier because I failed to double check to see if software messed up

Haidiye00, 2025-05-08:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

I'm very sorry, can your video capture the entire software? Thank you

Haidiye00, 2025-08-04:
Hello, please use the latest version. We are currently unable to reproduce it. If there are still issues, could you please provide the GIF operation reproduction process? Thank you