bambulab/BambuStudio #6721: Unable to connect to printers after update

Fredericphillips, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windown 11 PRO 

### Additional system information

Device name	DESKTOP-3PU6BCN
Processor	13th Gen Intel(R) Core(TM) i7-13700HX   2.10 GHz
Installed RAM	80.0 GB (79.7 GB usable)
Device ID	39CCF5DF-A12F-4CB2-8B63-DBB15EC62E64
Product ID	00330-80000-00000-AA458
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

X1C and H2D

### How to reproduce

Printer selection, you click on the printers (X1C AND H2D). Press confirm, and nothing happends. They dont show up. 

### Actual results

Nothing, the printers dont come up. 

### Expected results

I should see the printers... 

### Project file & Debug log uploads

Not project byt system problem 

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-27:
@Fredericphillips  Thank you for your feedback. Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

Fredericphillips, 2025-04-27:
Hello boboyu, thanks for getting back to me. I unistalled the latest version and installed the previouse version as i needed to get some prints done. I will do a Screen video grab once the print is finished in 2/3 hours

BoBoYu-BamBu, 2025-04-27:
@Fredericphillips OK. Looking forward to your next reply.

MrdUkk, 2025-04-27:
i encountered this issue too after updating to latest BS.
LAN only mode is only possible for me now after upgrade... very sad.... QA is not perfect as i see

Fredericphillips, 2025-04-27:
> [@Fredericphillips](https://github.com/Fredericphillips) Thank you for your feedback. Could you please provide a screen recording or a screenshot to describe the problem? This will facilitate our reproduction and resolution of the issue.

It looks like unistalling and re installing did the trick, no problem now. Works perfect.

closed 2025-04-27 by Fredericphillips