bambulab/BambuStudio #6722: unable to control printer fans from bambu studio

IuseArchBTW69, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab x1c

### How to reproduce

I am using the latest beta firmware on my printer
click on the fan option in bambu studio

### Actual results

a pop up appears with the text "Cooling mode is suitable for printing PLA/PETG/TPU materials and filters the chamber air" with no option to control the fans. a similar issue is occurring on the Bambu handy app where whenever i set the fan speed on any of the fans to %100 it limits it to %80 or %60 

### Expected results

I can click the fan option and have control over the fans

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-27:
Same as https://github.com/bambulab/BambuStudio/issues/6598#issuecomment-2819814418

shuizhiguo, 2025-04-28:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

TinyMito, 2025-04-30:
Someone posted this also:
https://www.reddit.com/r/BambuLab/comments/1kb6tj6/x1c_latest_firmware_broke_the_fan_controls_in_bs/

TinyMito, 2025-04-30:
Fan control is available on P1S but not X1C.

kle7rx, 2025-05-03:
Same problem. X1C printer.

[image]

i3dm, 2025-05-11:
same problm here

BoBoYu-BamBu, 2025-05-29:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

Please follow this guideline to create the after-sales order form. They will handle it.