bambulab/BambuStudio #6723: Flush to object not working

stonesolves91, 2025-04-27:
### Bambu Studio Version

01.07.00.00

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. select object
2. click "other" tab under advanced settings
3. scroll down and select all three "flush" options to create "purge print"

### Actual results

Every time i try to make a "poop print" where i flush to object, it is treating the object as a secondary priming tower. Can you please have someone reach out?

### Expected results

an object created from the purged filiment during color changing process

### Project file & Debug log uploads

i dont have this at the moment

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-28:
@stonesolves91  Could you record a screen or take a screenshot to clearly describe your problem?

stonesolves91, 2025-04-29:
system fails to upload every file i try. can i send via discord? system doesnt even allow me to upload zipped file