bambulab/BambuStudio #6725: Bambu Studio 'Home' is BLANK for v.2.0.3.54

rrrichardson, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1C

### How to reproduce

Install v .3.54, open it; home screen is blank.

### Actual results

Unusable / blank home screen

[image]

### Expected results

Loads a home screen.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-28:
@rrrichardson  Thank you for your feedback. Could you try again with a different network? If it still doesn't work, please provide the log file for us to analyze the problem.

rannski, 2025-04-29:
i don't think it's network related. even if not connected to the network, changing the BambuStudio root folder to any other name causes the flushing volumes to go blank

rannski, 2025-04-29:
how do i get the log file?

we-rebel-scum, 2025-05-28:
> how do i get the log file?

On Windows it is:
Menu -> Help -> Show Configuration Folder
Open the "log" folder and find the most recent file.

I too am having this issue on the same build number. My logs are attached. For quick reference, this seems to be where it's failing in the logs, which repeats a few times before giving up and producing the blank home screen:
2025-05-28 15:02:40.593279[Thread 0x0000329c]:create the main window
2025-05-28 15:02:40.695403[Thread 0x0000329c]:WebViewEdge::SetUserAgent: create web view failed.
2025-05-28 15:02:40.701403[Thread 0x0000329c]:WebView::CreateWebView::<lambda_7f7fefeaf648454a6c8b285d39d8fc33>::operator (): begin to add script message handler for wx.
2025-05-28 15:02:40.701403[Thread 0x0000329c]:WebView::CreateWebView::<lambda_7f7fefeaf648454a6c8b285d39d8fc33>::operator (): finished add script message handler for wx.
2025-05-28 15:02:40.701403[Thread 0x0000329c]:WebViewEdge::SetUserAgent: create web view failed.

[Bambu_studio_log.zip](https://github.com/user-attachments/files/20495618/Bambu_studio_log.zip)