bambulab/BambuStudio #6726: Please remove anti-debugging code in BBL network plugin

Noisyfox, 2025-04-27:
Since 1.10.2 the BBL network plugin ships with debugger detecting code that will terminate the application when a debugger is attached. This makes it impossible to run the slicer from Visual Studio or any dev tools with debugger attached, which means there is no way to do any meaningful development as long as you own a BambuLab printer.

Please, stop this. No one is interested in reverse engineering your network plugin. Everything you tried to hide has already been leaked in countless different way. The certificate has been extracted from BambuConnect, your network protocol has been figured out for a long time now, and any attempt of hiding things by encryption will be broke as long as the dll is shipped to end user and executed on their machine. All you are doing here is preventing people from contributing to this "open-sourced" project. And think about this, now people have to crack your network plugin just so that we could develop the reset of the project. Why it has to be like this? I have absolutely no interest in what you tried to hide in the dll, just let me run the app directly from Visual Studio.

DenisVth, 2025-04-28:
+1. Stop killing other slicer, please!

CodyPrince, 2025-04-30:
Is this still open sourced? Kind of sounds like they closed it.

awdev1, 2025-04-30:
+1 Provide an SDK to others in this case
its my printer.
i bought it
you are using my wifi
you are using my filament
I should have freedom over it all

devjj, 2025-04-30:
Truly wild behavior. My X1C is my first Bambu machine and absolutely my last, too, so long as this crap stands.

jk, 2025-04-30:
Maybe it’s time for a complete open sourced, separate firmware.

JadElClemens, 2025-04-30:
So much of Bambu's recent moves are in total opposition to the spirit of the 3D printing community. They're damn good printers, but this nonsense outweighs all of that quality.

Fisheye3D, 2025-04-30:
> Maybe it’s time for a complete open sourced, separate firmware.

There is one. x1plus firmware

awdev1, 2025-04-30:
> > Maybe it’s time for a complete open sourced, separate firmware.
> 
> There is one. x1plus firmware

I wonder if there is one for the stock P1S

Orel-A, 2025-04-30:
Is this the `%AppData%\BambuStudio\plugins\bambu_networking.dll` file? Maybe someone could analyze it and figure it out. I don't think it's packed with obfuscation software

Noisyfox, 2025-04-30:
> Is this the `%AppData%\BambuStudio\plugins\bambu_networking.dll` file?

Yes.

joelhaasnoot, 2025-04-30:
> So much of Bambu's recent moves are in total opposition to the spirit of the 3D printing community. They're damn good printers, but this nonsense outweighs all of that quality.

While I agree with you completely, I think the reason for all this is likely more attacking other competitors that are reusing Bambu firmware and code to get a leg up in the market and/or provide cheaper products. This isn't a fight against users for Bambu, it's a fight against competitors.

johnny-bit, 2025-04-30:
> This isn't a fight against users for Bambu, it's a fight against competitors.

Fight against competitors IS fight against users. Monopolies aren't good for users ;) Also - competitors aren't using bambu firmware, all I'm seeing is klipper or some totally different solutions. Leaving the firmware side of that there's code - and the code of Bambu Studio is AGPL licenced which means it HAS to be opensource. So that leaves only the network protocol. And the network protocol is certainly not used by others while it's now used as a anti-developer solution :/ very not cool.

Orel-A, 2025-04-30:
> I think the reason for all this is likely more attacking other competitors that are reusing Bambu firmware and code to get a leg up in the market and/or provide cheaper products

What are you talking about? Do you even know what this DLL file's purpose is? It's a simple API that lets the slicer send print jobs to Bambu printers. Furthermore, the firmware code resides on the printer itself, not on the end user's personal machine. So debugging / reverse engineering this file wouldn't give any benefit to competitors.

Orca developer doesn't care if this DLL file is a closed source binary file. But the fact of the matter is that Orca loads this file into its executable memory and **this DLL file shuts down OrcaSlicer** in development, hindering Orca team to continue its support for Bambu newer firmwares. Is this intentional? Probably not, but it is indeed suspecious behaviour. At least Bambu's developers should have been aware of the bad impact this will cause to Orca, but they chose to ignore it.

SufficentMuffin, 2025-04-30:
+1 Stop this SH*T!

psiberfunk, 2025-04-30:
This continues Bambu’s recent unfortunate behavior of being an open source community leach.  I’m not sure if this is just a cultural gap with China where open source development is relatively uncommon, an utterly daft mess of a business decision, or actually malicious .   I tend to think it’s the first two , but it looks bad either way .

CCS86, 2025-04-30:
I was going to purchase an H2D, but will not until Bambu resolves this rift.

jarombouts, 2025-04-30:
Guys/gals/entities at Bambu. I dropped many hundreds of $$$ on your gear, and I do not like where this is going. In poetic form:

---

it looks good

it prints well

BUT DONT FUCKING GO THIS WAY

YOU HAVE A NICE PRODUCT

DONT LISTEN TO FUCKING MORONS WITH AN MBA THAT GIVE A SPIEL ON "customer lifetime value" AND "monthly recurring revenue" AND "strategic limitations in user facing APIs"

IT IS DUMB 

YOU ARE NOT ORACLE

YOU ARE SERVING THE MAKER COMMUNITY NOT SUITED UP CORPO LEECHES WITH THREE MISTRESSES AND A CIALIS ADDICTION THAT YOU CAN STRATEGICALLY wine and dine SO THEY BUY MORE OF YOUR PRODUCT

FUCK

---

Thank you.

psiberfunk, 2025-04-30:
> > > Maybe it’s time for a complete open sourced, separate firmware.
> 
> > x1plus firmware
> 
> It would be far better if Bambu Labs opened up their software ecosystem and stopped being protectionists to the detriment of their users

I think it's clear that they're only willing to pay lip service to this type of thing when they get publicly embarrassed into it to save face.  The mentality is still very closed internally, it would seem.  It is sad that they only see open-source as a PR thing to be "handled" and paid lip service instead of actually abided by and supported.

psiberfunk, 2025-04-30:
> BUT DONT FUCKING GO THIS WAY

... this type of discourse is unlikely to be productive in dissuading them.  Please use cogent arguments instead of ranting at them, which they will use as a reason to ignore this issue.

jarombouts, 2025-04-30:
> [@jarombouts](https://github.com/jarombouts)
> 
> > BUT DONT FUCKING GO THIS WAY
> 
> ... this type of discourse is unlikely to be productive in dissuading them. Please use cogent arguments instead of ranting at them, which they will use as a reason to ignore this issue.

You're not wrong, haha.

awdev1, 2025-04-30:
> Maybe it’s time for a complete open sourced, separate firmware. 

Perhaps a self hosted cloud? Maybe we can host our own coordination server if Bambu allows us at all. Seeing that they want to expand into business operations it would be a good idea. As uploading things to the cloud isn't secure for patented things sometimes

David-Everlasting-Media, 2025-05-01:
I purchased a BL printer because they seemed to be going the opposite direction of this. This hearts my heart. The bait and switch was predicted by many, heeded by few, and now look where we are. 

Bambu Lab, if you care at all about the makersphere, about your customers and their freedom to choose, fix this.

awdev1, 2025-05-01:
> I purchased a BL printer because they seemed to be going the opposite direction of this. This hearts my heart. The bait and switch was predicted by many, heeded by few, and now look where we are. [...]

When I saw the interview the CEO had with CNC Kitchen, I was immediately concerned for the OSS Community. The CEO clearly had no intention of doing the right thing and open-sourcing things.  Yet, I was forced to buy it as our robotics team needed a reliable quality printer, and Bambu Lab was the only good option at the time.

Prazgreen1, 2025-05-11:
Bambu, I'll just say this:
You actively benefit from open source development from Orca Slicer in Bambu Studio. There used to be so much upstream and downstream development that actively benefited both projects and the end user with no damage to Bambu. Now, I am fine if cloud mode is restricted under Bambu Studio as that is a reasonable compromise since it is your own servers. But, there is no reason why you should restrict this development because it hurts your end users and you yourself since less people are buying Bambu and also because you rely on this proper development. You are actively saving money and thus helping your bottom line by supporting the development of Orca Slicer. Let's be honest, the 3D printer industry is one where brand loyalty isn't really given. We choose the machines that make the most sense for our needs in that machine. A unified slicer for all machines doesn't hurt Bambu due to the competition it encourages but rather benefits Bambu as many users have multiple machines from multiple brands. There are many printer owners who used to juggle many slicers and were fine with it and still are but prefer the unified platform. SoftFever is a great developer, and the mere fact that you can save significant money and make more by supporting OrcaSlicer is a mere fact. Furthermore, by stopping users from contributing to BambuStudio, you are preventing end-users from submitting pull-requests with features that benefit other end-users without you needing to spend pretty much any money paying for such development. There is no logical reason for raising the walls on this garden. I mean, the market has gotten much more competitive to the point where even the A1 Mini is now competing on price with the Creality Hi. If you want to stay relevant, it is imperative for you to actively support the end-users and thus the community you are designing machines for. The entire security update is a bundle of lies because the only actually secure way to implement security is public key encryption. Just be forthcoming.
Also, do not block users from utilizing features in their machines through firmware updates without releasing the proper tools needed. Bambu Connect is still in beta and has very little features with a Linux version being completely unavailable. Now, Bambu Connect is a welcome addition in my eyes if it were just another connection standard or something that was required just for cloud use. The main issue in my eyes is the utter lack of concern when it comes to actually letting the printer have proper development around it. The entire consumer 3D printing industry is built upon open-source development. Removing that destroys your reputation and actively makes your own products worse.

UltraBlackLinux, 2025-05-14:
I think we've reached the point where I think it's pretty safe to say if you want to **own your printer** you should just avoid bambu altogether.
Since the release of the A1 and -mini other companies have caught up in terms of speed and print quality, without relying on non-standard gcode and closed source networking plugins.

Who would have thought that the apple of 3d printing is just as bad as the real apple...

skl111, 2025-05-16:
Oh you most definitely should avoid them since they will remote brick your printer if you choose to mod your printer.

> We've reached the point where I think it's pretty safe to say if you want to **own your printer** you should just avoid bambu altogether. Since the release of the A1 and -mini the competition has caught up in terms of speed and print quality, without relying on non-standard gcode and closed source networking plugins. Never update the firmware on your bambu printer, because the issues that await clearly outweigh the features you gain.
> 
> Who would have thought that the apple of 3d printing is just as bad as the real apple...

eeverman, 2025-06-11:
Ugh.  This is so disappointing.  I'll be waiting before I consider the HD2.

UltraBlackLinux, 2025-06-12:
@eeverman forget BBL altogether. Get something like a prusa instead. Bambu will never again return to what they were.

Frozander, 2025-07-12:
Never getting any other bambu printer again. Disgusting behaviour that they are ghosting this issue.

jk, 2025-07-12:
Same here. Went with a new Prusa instead of a Bambu. Wouldn‘t have done it otherwise. So that’s the result of their politics. Also they miss out of me buying their filament for my existing Bambu, so it backfired on both ends.

tsmith35, 2025-07-27:
We're buying 3D printers for the technical staff at work, and I was asked to help decide which printers we would get. Despite my owning a half-dozen Bambu Lab printers, my recommendation was to go with Creality at this time. We also have a Prusa Core One coming in soon. The present situation with BBL's decision to lock down slicer selection eliminated BBL printers from consideration.

BBL burned that bridge on their own. I have no idea what BBL is gaining from their bizarre decision, but I know what they're losing.

Nellieflem, 2025-07-31:
I was looking so forward to buying the HD2.  I will buy one if this all goes away and back to what it was.  I have four BBL printers and can never update again.  Breaks my heart.

jacobcxdev, 2025-08-04:
I'm not sure if they've reversed it or what, but after updating dependencies and fixing a bunch of build errors, I'm able to build and debug on macOS using JetBrains Rider, LLDB, etc. Perhaps their anti-debugging measures only work on Windows, but it seems to me that development is still possible.

Noisyfox, 2025-08-04:
> I'm not sure if they've reversed it or what, but after updating dependencies and fixing a bunch of build errors, I'm able to build and debug on macOS using JetBrains Rider, LLDB, etc. Perhaps their anti-debugging measures only work on Windows, but it seems to me that development is still possible.
> 
> [jacobcxdev@71c022e](https://github.com/jacobcxdev/BambuStudio/commit/71c022e2dd3ab744c466dfd227c364e723f7f031)

Have you enabled the bambu network plugin?

Noisyfox, 2025-08-04:
> I'm not sure if they've reversed it or what, but after updating dependencies and fixing a bunch of build errors, I'm able to build and debug on macOS using JetBrains Rider, LLDB, etc. Perhaps their [...]

Hmmmm you're right actually. The latest plugin does work even if debugger is attached. Thanks for the info. I'm gonna play around with it a bit more and post back later.

Prazgreen1, 2025-08-04:
Well, that hopefully is a step in the right direction. I'd just love to see the bambu handy app to support lan mode.

Noisyfox, 2025-08-04:
I‘ve tried that on Orcaslicer, and it did work. Gonna do more testing.

psiberfunk, 2025-08-04:
> Well, that hopefully is a step in the right direction. I'd just love to see the bambu handy app to support lan mode.

Extremely unlikely to ever happen.  They want you on their cloud.

Noisyfox, 2025-08-04:
https://github.com/SoftFever/OrcaSlicer/pull/10320
So far so good.

Prazgreen1, 2025-08-05:
> > > Maybe it’s time for a complete open sourced, separate firmware.
> > 
> > 
> > There is one. x1plus firmware
> 
> I wonder if there is one for the stock P1S

They've gotten Klipper to run on the P1S

Prazgreen1, 2025-08-05:
probably wouldnt work stock though, but it's better than what I have with the A1 Mini., I just keep that on old firmware.

Noisyfox, 2025-08-08:
Closed since the new plugin allows debugger. Hopefully I don't have to reopen this in the future.

Noisyfox, 2025-08-27:
> [@Noisyfox](https://github.com/Noisyfox), I saw the new alpha appears to support the H2D and newer FW. So, for those of us holding out on updating X1C firmware, can we go ahead and update without worry? Or this this a precarious situation at the moment?

You will loose cloud print from 3rd party slicer if you use latest firmware.