bambulab/BambuStudio #6727: Chamber Cooling causes PETG printing failures on H2D – Manual fix needed / Slicer option missing

Gh3tt0b0y, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

all OS

### Additional system information

# Chamber Cooling causes PETG printing failures on H2D – Manual fix needed / Slicer option missing

## Description

Hi Bambu Lab Team,

after watching a detailed video about PETG printing issues on the H2D and consulting ChatGPT to analyze the problem more precisely, I would like to report the following:

---

## 🛑 Problem

- When starting a print on the **H2D**, the **chamber cooling fans (P2, P3)** are automatically enabled.
- This blows cold air into the chamber, leading to:
  - **Overextrusion**
  - **Premature filament solidification**
  - **High nozzle pressure**
  - **Extruder clog errors** and **ultimately print failures**.

This happens even when printing in **normal room temperatures (20–22 °C / 68–72 °F)** where chamber cooling is unnecessary for most materials.

Currently, there is **no option in Bambu Studio to control chamber cooling behavior per material**, only globally via printer firmware or G-code edits.

---

## 📦 Officially Supported Materials on H2D

According to official specifications, the H2D supports:

- PLA
- PLA-CF
- PETG
- PETG-CF
- TPU
- PVA
- PA (limited support for PA/PA-CF)

Most of these materials, especially **PLA, PETG, and TPU**, **do not benefit** from active chamber cooling at standard ambient temperatures and actually **suffer from it**.

---

## 📹 Video Reference

A video clearly demonstrates the problem and the manual workaround by disabling chamber cooling after starting the print:

🔗 **Video Link:** [PETG Printing Problem with Chamber Cooling – Watch here](https://youtu.be/sYN-dwEzZQ8)

---

## 🛠 Workaround for H2D Users

Until Bambu provides an official fix, the following workaround can reliably solve the problem:

### 1. Modify the Start-GCode:

Replace the "air duct" section with:

```gcode
;==== set airduct mode ==== 

; - Start Chamber Fix Code -
; Always disable chamber fans to prevent PETG/PLA print issues
M145 P0 ; set airduct mode to cooling
M106 P2 S0 ; auxiliary fan OFF
M106 P3 S0 ; chamber fan OFF
;M140 S0 ; (optional: do not stop heatbed)
;M1002 gcode_claim_action : 29 ; (optional: disable chamber wait)
;M191 S0 ; (optional: skip chamber wait)
;M142 P1 R35 S40 U0.3 V0.5 ; (optional: disable autocooling)
; - End Chamber Fix Code -

;==== set airduct mode ==== 

### Printer

Bambu Lab H2D

### How to reproduce

1. Slice a PETG or PLA model using Bambu Studio for the H2D printer (standard profiles, no manual G-code changes).
2. Start the print on the H2D as usual.
3. Observe that after the printer starts heating, the chamber cooling fans (Auxiliary Fan and Chamber Fan) are automatically turned ON.
4. Let the print continue without manually disabling the chamber cooling.
5. Notice the following symptoms during the print:
   - Overextrusion
   - Irregular filament flow
   - Clogging at the extruder
   - Possible print failure and error messages (e.g., filament clog detected).
6. Optionally, repeat the same sliced G-code but **immediately disable the chamber cooling manually via the printer’s control panel after starting the print**.
7. Observe that with chamber cooling OFF, the print completes successfully without any issues.

---

### Actual results

- The chamber cooling fans (Auxiliary Fan and Chamber Fan) turn ON automatically at the start of the print.
- Cold air is blown into the build chamber, even though ambient room temperature is normal (around 20–22 °C).
- The filament (especially PETG or PLA) cools prematurely, causing:
  - Overextrusion artifacts
  - Filament solidifying before proper layer adhesion
  - Increased pressure inside the nozzle
  - Extruder clog detection and resulting print failure
- The printer may stop the print and display an error message (e.g., filament clog or extrusion error).

---

### Expected results

## 🎯 Expected Results

- The chamber cooling fans (Auxiliary Fan and Chamber Fan) should **remain OFF** when printing with materials like PETG, PLA, TPU, or PVA at normal ambient temperatures (around 20–22 °C).
- The printer should **not automatically activate chamber cooling** unless it is specifically needed (e.g., high chamber temperature, high-temp materials like PA or PC).
- PETG, PLA, and similar materials should print reliably without:
  - Overextrusion
  - Premature filament cooling
  - Nozzle pressure buildup
  - Extruder clogging errors
- The user should not be required to manually intervene to disable chamber cooling after the print starts.
- Ideally, the slicer or firmware should provide a **per-material option** to control chamber cooling behavior automatically.

---

### Project file & Debug log uploads

## 🛠 Project File & Debug Log Uploads

- Since the issue is not related to a specific model or project file, any standard PETG or PLA print sliced with default H2D profiles in Bambu Studio can reproduce the behavior.
- The problem occurs independently of specific G-code settings, as it is caused by the printer's automatic chamber cooling behavior at the start of a print.
- Debug logs are not necessary to demonstrate this issue, because it can be directly observed by starting any PETG or PLA print and monitoring the chamber fans' behavior.
- However, if needed, I can provide:
  - Example project files (.3mf) sliced with default PETG profiles.
  - Short video recordings showing the chamber cooling behavior and the resulting print issues.
  - Debug logs from the printer after a failed print, if required for further analysis.

Please let me know if you would like me to upload any additional files for investigation.

### Checklist of files to include

- [x] Log file
- [x] Project file

Gh3tt0b0y, 2025-04-27:
`;===== machine: H2D =========================
;===== date: 20250418 =====================

;M1002 set_flag extrude_cali_flag=1
;M1002 set_flag g29_before_print_flag=1
;M1002 set_flag auto_cali_toolhead_offset_flag=1
;M1002 set_flag build_plate_detect_flag=1

M993 A0 B0 C0 ; nozzle cam detection not allowed.

M400
;M73 P99

;=====printer start sound ===================
M17
M400 S1
M1006 S1
M1006 A53 B9 L99 C53 D9 M99 E53 F9 N99 
M1006 A56 B9 L99 C56 D9 M99 E56 F9 N99 
M1006 A61 B9 L99 C61 D9 M99 E61 F9 N99 
M1006 A53 B9 L99 C53 D9 M99 E53 F9 N99 
M1006 A56 B9 L99 C56 D9 M99 E56 F9 N99 
M1006 A61 B18 L99 C61 D18 M99 E61 F18 N99 
M1006 W
;=====printer start sound ===================

;===== reset machine status =================
M204 S10000
M630 S0 P0

G90
M17 D ; reset motor current to default
M960 S5 P1 ; turn on logo lamp
G90
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
M73.2   R1.0 ;Reset left time magnitude
G29.1 Z{+0.0} ; clear z-trim value first
M983.1 M1 
M901 D4
M481 S0 ; turn off cutter pos comp
;===== reset machine status =================

M620 M ;enable remap

;===== avoid end stop =================
G91
G380 S2 Z22 F1200
G380 S2 Z-12 F1200
G90
;===== avoid end stop =================

;==== set airduct mode ==== 

{if (overall_chamber_temperature >= 40)}

    M145 P1 ; set airduct mode to heating mode for heating
    M106 P2 S0 ; turn off auxiliary fan
    M106 P3 S0 ; turn off chamber fan

{else}
    M145 P0 ; set airduct mode to cooling mode for cooling
    M106 P2 S178 ; turn on auxiliary fan for cooling
    M106 P3 S127 ; turn on chamber fan for cooling
    M140 S0 ; stop heatbed from heating

    M1002 gcode_claim_action : 29
    M191 S0 ; wait for chamber temp
    M142 P1 R35 S40 U0.3 V0.5 ; set chamber autocooling
    M106 P2 S0 ; turn off auxiliary fan
[... 404 lines cut ...]

M400
;M73 P99

M993 A1 B1 C1 ; nozzle cam detection allowed.

{if (filament_type[initial_no_support_extruder] == "TPU")}
M1015.3 S1;enable tpu clog detect
{else}
M1015.3 S0;disable tpu clog detect
{endif}

{if (filament_type[initial_no_support_extruder] == "PLA") ||  (filament_type[initial_no_support_extruder] == "PETG")
 ||  (filament_type[initial_no_support_extruder] == "PLA-CF")  ||  (filament_type[initial_no_support_extruder] == "PETG-CF")}
M1015.4 S1 K1 H[nozzle_diameter] ;enable E air printing detect
{else}
M1015.4 S0 K0 H[nozzle_diameter] ;disable E air printing detect
{endif}

M620.6 I[initial_no_support_extruder] W1 ;enable ams air printing detect

M211 Z1
G29.99

`

hotellonely, 2025-04-28:
This is mainly a bambu firmware issue, not a Bbstudio issue. In short Bambu's firmware don't have a smart temperature control and it's rather naive on the cooling strategy. Should submit a ticket to support instead. Ideally we want smart temperature control that keeps the chamber temperature to the ideal range dynamically. But what Bambu is doing right now is just following a set rule regardless of your actual chamber temp and room temp.

gaolegao-lx, 2025-04-29:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

P-C-R, 2025-05-16:
+1 for better chamber control. Running the heater fan at 100 percent is not needed tbh

closed 2025-05-29 by BoBoYu-BamBu

hotellonely, 2025-05-29:
Is this completed? @BoBoYu-BamBu