bambulab/BambuStudio #6730: 新版本中开启擦拭塔会导致在非换料层过量回抽

Von-Clausewitz, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

masOS 15.2

### Printer

P1S/H2D

### How to reproduce

新版本中开启擦拭塔，在非换料层打印完擦料塔之后，会有一个过量回抽
同一模型默认材料/打印机设置，通过阅读gcode发现，旧版（指2.0之前的版本）在打印完擦拭塔后的回抽值会跟随材料/打印机设置里的回抽值（例如设定0.8实际为0.76，设定0.4实际为0.38），而新版（指2.0到最新的稳定版2.0.3）会有一个不受材料/打印机设置影响的固定回抽值，约为1.9

### Actual results

由于这个回抽值过大，会导致空气进入喷嘴，之后的打印会出现气泡，进而引发外墙缺陷

### Expected results

不该有这么大的回抽

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

1754135043, 2025-04-28:
我还以为就我有问题，希望可以选择使用回旧版本的擦料塔吧

hotellonely, 2025-04-28:
Yeah that's a quite annoying issue. Need a fix asap.

hotellonely, 2025-04-28:
[image]

As shown above. That E-1.89998 step is kinda confusing.

1754135043, 2025-04-28:
> 是的，这是一个非常烦人的问题。需要尽快修复。

When version 2.0 came out, I already felt there were issues and provided feedback, but it’s still not resolved, so I had to revert to a previous version.

zhimin-zeng-bambulab, 2025-05-06:
Thanks for reporting this issue, we will fix it in the next version.

zhimin-zeng-bambulab, 2025-06-05:
The new version ([v02.01.00.59](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59)) has fixed this problem, you can have a try.

closed 2025-06-26 by zhimin-zeng-bambulab