bambulab/BambuStudio #6731: Application suddelny closing

IPerino, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro (10.0.19045)

### Additional system information

Intel Core i9-9880H 2.3GHz
64GB RAM

### Printer

X1 Carbon w AMS

### How to reproduce

Not possible to reproduce, when working with Bambu studio, taking some moments to parametrize, add support, etc. then, suddenly it shut off, it closes down without any sign. It has happened more than 10 times with different projects.
Once the application is open again, it offers retake the job where it was. However, the it has to be sliced again.

[log.zip](https://github.com/user-attachments/files/19932014/log.zip)

### Actual results

NA

### Expected results

NA

### Project file & Debug log uploads

log files attached

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-28:
@IPerino Sorry, could you provide me with the 3mf file?

BoBoYu-BamBu, 2025-04-28:
@IPerino  Was it a crash during the slicing process? Or did it crash during other operations?

IPerino, 2025-04-29:
Sorry for the delay in the response.
It has happened many times and not necesarily with a project file. When I prepare and print the part, not always make a project and save it. 
It does not close when slicing, it closes suddenly when you are just watching the part, thinking on the parameter, etc. The dump file is there in the attachment. log.zip
I didnt see it happening in the previous version I had installed. Sorry, not sure what version was. I can check the installer if you want.