bambulab/BambuStudio #6732: Flow Rate calibration: not possible to overwrite existing presets

vanowm, 2025-04-27:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 x64

### Printer

X1C

### How to reproduce

Do Flow Rate calibration, when asked for a name to save, type a name of already existing preset

### Actual results

Error popups saying the name is the same as existing preset

### Expected results

It should warn you that a preset with this name already exists and if you want to overwrite it.

### Project file & Debug log uploads

.

### Checklist of files to include

- [x] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-04-30:
@vanowm  We do not allow overwriting system presets, only user-defined presets. What type of preset do you want to overwrite?

vanowm, 2025-04-30:
Overwriting user presets

zhimin-zeng-bambulab, 2025-05-06:
Could you take a screenshot of your error message? And could you share your log files?

vanowm, 2025-05-27:
[image]

[debug_Tue_May_27_14_21_05_45043.log.zip](https://github.com/user-attachments/files/20469415/debug_Tue_May_27_14_21_05_45043.log.zip)