bambulab/BambuStudio #6733: 新版本的擦料塔对比旧版本的会使模型质量变差

1754135043, 2025-04-28:
### Bambu Studio Version

v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

Bambu P1SC

### How to reproduce

昨天晚上和Bambu Handy的技术人员测试了一晚上，测试内容是关于一个模型不开擦料塔，以及开启新版擦料塔跟旧板擦料塔的模型质量对比。使用的打印工艺均相同

### Actual results

经过测试发现，开启旧版本擦料塔与不开启擦料塔的模型质量几乎一致，而开启新版擦料塔的话模型会出现材料粘贴、大量气泡以及擦料塔缺料等...
我添加了一个文件夹图片，图片1是我打印大模型擦料塔的情况，图片2是与技术人员交流的聊天截图，图片3~6（视频）是一个模型开启新版本擦料塔的外观质量拍摄，图片7~8是同一个模型（小船），使用相同打印工艺（0.12mm High Quality @BBL X1C ），开启新版本与旧版本擦料塔的对比图，上方的是新版本，下方的是旧版本。图片9是同一个模型（https://makerworld.com.cn/zh/models/1162775-fa-na-ke-lr-mate-200idgong-ye-ji-qi-ren#profileId-1227417），使用相同打印工艺（0.12mm High Quality @BBL X1C ），开启新版本与旧版本擦料塔的对比图，上方的是新版本，下方的是旧版本。

[材料.zip](https://github.com/user-attachments/files/19932500/default.zip)

### Expected results

我希望新版本的切片软件里面可以选择使用旧版本的擦料塔

### Project file & Debug log uploads

[材料.zip](https://github.com/user-attachments/files/19932507/default.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

1754135043, 2025-04-28:
刚才我参考了一下其他人的意见，看来都有类似的问题，我这也确实出现了有大量气泡的问题

[image]

zhimin-zeng-bambulab, 2025-04-30:
和 https://github.com/bambulab/BambuStudio/issues/6730 这个问题相关
感谢您报告这个问题！我们将在下个版本修复这个问题。

JohnSmithCYM, 2025-05-30:
我们在内部已经尝试修复了该问题，您能否提供一个稳定出现问题的3mf供我们验证？