bambulab/BambuStudio #6736: The support part is not normal when slicing, did this update break the support?

TaoWolf, 2025-04-28:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: INTEL i7 13
Memery: 32G
GPU: Rtx4090

### Printer

Bambu P1S

### How to reproduce

The support that comes out of the slice is not normal.
1. Select Generate only on the print plate, it is still generated inside the object.
2. There is a fault in the middle of the support, which cannot be printed normally!

### Actual results

The support that comes out of the slice is not normal.
1. Select Generate only on the print plate, it is still generated inside the object.
2. There is a fault in the middle of the support, which cannot be printed normally!
3. The support surface is set up with 5 layers, but every time the support is generated, there is no support surface, which makes it difficult to dismantle and the effect is not good!
Find a few objects with a lot of support, cut them and you'll see.

### Expected results

The support that comes out of the slice is not normal.
1. Select Generate only on the print plate, it is still generated inside the object.
2. There is a fault in the middle of the support, which cannot be printed normally!
3. The support surface is set up with 5 layers, but every time the support is generated, there is no support surface, which makes it difficult to dismantle and the effect is not good!
Find a few objects with a lot of support, cut them and you'll see.

### Project file & Debug log uploads

[image]

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hotellonely, 2025-04-28:
try modify the layer height of the support and see if using the same layer height in both the print and the support helps.

BoBoYu-BamBu, 2025-04-28:
@TaoWolf  Sorry, could you provide me with the 3mf file?