bambulab/BambuStudio #6739: Flushing volumes tab is blank when bambustudio application folder name is other than the default

rannski, 2025-04-28:
### Bambu Studio Version

happens in 2.0.2 and 2.0.3

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sequioa 15.4.1

### Additional system information

I usually name my BS application folder according to version i.e. BambuStudio 2.0.1. However, with 2.0.2 and 2.0.3, flushing volumes tab is blank. When I used the default folder name, it was fixed. Please look into it.

### Printer

P1S

### How to reproduce

Rename the application folder other than the default

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2025-05-10:
@rannski This is related to the sandbox mechanism of macOS. You can first remove the version information from the application folder name. We are currently analyzing this issue.

rannski, 2025-05-10:
to add, it's ok on BS v2.0.1

XunZhangBambu, 2025-06-03:
@rannski  This has been fixed in our latest version 02.01.00.59