bambulab/BambuStudio #6740: FLush Volumes Not Adjustable

ZIIIWAY, 2025-04-28:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS  14.5

### Printer

h2d

### How to reproduce

[image]
Clicked the flushing volume menu and there are no option, simply a white window

### Actual results

[image]
Clicked the flushing volume menu and there are no option, simply a white window

### Expected results

[image]
Supposed to be options

### Project file & Debug log uploads

[Gyro+Card+Fidget.3mf.zip](https://github.com/user-attachments/files/19939661/Gyro%2BCard%2BFidget.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

XunZhangBambu, 2025-05-10:
@ZIIIWAY  Same as https://github.com/bambulab/BambuStudio/issues/6282. When installing, you need to drag Bambu Studio into the application folder and do not change the name of the application folder.

XunZhangBambu, 2025-06-03:
@ZIIIWAY This has been fixed in our latest version 02.01.00.59

closed 2025-06-03 by XunZhangBambu