bambulab/BambuStudio #6742: Color Scheme Dropdown

jojomanzo, 2025-04-28:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Window 11

### Printer

PS1

### How to reproduce

Go to PREVIEW or SLICE
When  you see the SLICING RESULT window popup
Try to click on the dropdown to change Color Scheme
I click on it and it does not dropdown for a few seconds (4 or 5) then when  use the mouse or my finger to try to select a new one, the dropdown goes "behind" the pop up window, then fades out.  I need to click the drop down, and try to hold the mouse over the selection for a LONG TIME until it turns green. Then sometimes it will let me select it. This has never happened with the older versions, please fix.

### Actual results

Drop down goes away

### Expected results

I should be able to easily pick any item from the dropdown

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

Haidiye00, 2025-10-15:
Hello, please update the software to the new version.
Since you haven't replied for a long time, I will close the issue first. If there are still any issues, please record a GIF and upload it. Thank you

closed 2025-10-15 by Haidiye00