bambulab/BambuStudio #6744: TREE(AUTO) SUPPORT NOT WORKING AFTER LATEST UPDATE

emwolb16, 2025-04-28:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

BAMBU A1

### How to reproduce

i can add normal support but when i select tree there are no supports.  I'm using the same files I used before the update.

### Actual results

see above

### Expected results

see above

### Project file & Debug log uploads

[REAR BUMPER.zip](https://github.com/user-attachments/files/19944148/REAR.BUMPER.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-BamBu, 2025-04-29:
@emwolb16 Could you record a screen or take a screenshot to clearly describe your problem?

NothFie, 2025-04-29:
@emwolb16 The determination of 'small overhang' is a little different in Tree support and Normal support. These overhangs are removed in Tree support. I tried printing without support, the result wasn't very good. In future versions we will improve the detection of overhangs, for now you could close the option "remove small overhangs" or choose to use Normal support.

[image]

[image]